ArduPilot/ardupilot #5337: Compiler Error in Plane master

priseborough, 2016-12-06:
#### Issue details
Commit https://github.com/ArduPilot/ardupilot/commit/8d8775125a20031ade3b5fff9aa71e3e20c69050 appears to be causing a compiler error.

I used the following build  sequence:

git fetch origin
git checkout master
git reset --hard origin/master
git submodule update --init --recursive
./waf clean
./waf configure --board px4-v2
./waf --targets bin/arduplane --upload

And got the following error:

../../libraries/AP_Landing/AP_Landing.cpp: In member function 'void AP_Landing::setup_landing_glide_slope(const Location&, const Location&, const Location&, int32_t&)':
../../libraries/AP_Landing/AP_Landing.cpp:190:36: error: qualified-id in declaration before '(' token
void AP_Landing::init_start_nav_cmd(void)
^

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
px4-v2

#### Logs
N/A

tridge, 2016-12-06:
fixed in master by reverting 2 patches

closed 2016-12-06 by rmackay9

magicrub, 2016-12-06:
Doh! Thanks for the quick fix, Tridge. Sorry about that everyone!