ArduPilot/ardupilot #5342: Feature request - driver interface

night-ghost, 2016-12-06:
#### Issue details
Now callback function for register_periodic_callback()  is defined as FUNCTOR_TYPEDEF(PeriodicCb, bool) - so it returns bool. This allows to return only 2 states - OK and RemoveMe. But usually callback function want to return 3rd state - RescheduleMeAtNextTick. This status is intended for cases when callback function can't do its job - eg. failed to get a semaphore. Sample - bool AP_Compass_HMC5843::_timer() function: when it failed to get semaphore in line 264, it simply discards last values from compass. May be it will be better to get semaphore before reading from compass, and return RescheduleMeAtNextTick?

All is much  worse for barometer: code tries to get semaphore with flag HAL_SEMAPHORE_BLOCK_FOREVER (AP_Baro_MS5611.cpp line 293) - so if HAL don't support preemptive multitasking all will ends in Hal::panic(). The solution is the same - try to get the semaphore before read, and if failed return RescheduleMeAtNextTick.

#### Version
git

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

magicrub, 2016-12-06:
sounds useful

night-ghost, 2016-12-06:
Yes this is tested on my Revomini HAL. STM32 without any problems allows the scheduler to run at a frequency of 8 kHz, so that the extra delay will be only 0.125 microseconds - so not a full loop time

magicrub, 2016-12-06:
But you still have the horizon to deal with so it always needs to be handled no matter what the latency is.

night-ghost, 2016-12-06:
sorry I don't understand what you about. Oh, those Russian ...

peterbarker, 2021-07-30:
The HMC5483 drier has changed significantly, and now does hold the semaphore as suggested in here.

The example given for the new interface - unable to get a sem - is not great as we strongly prefer the take-blocking form now to avoid starvation.  We'd need a better example to change this interface.

closed 2021-07-30 by peterbarker