ArduPilot/ardupilot #5344: Copter: add support for HIL_GPS

rmackay9, 2016-12-06:
ROS's mavros sends "fake GPS" messages using the mavlink HIL_GPS message.
https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/mocap_fake_gps.cpp

We already support a "fake GPS" by setting the GPS_TYPE to "MAV" (14) but we only accept the GPS_INPUT message.  We should extend this mavlink gps driver to accept the HIL_GPS message as well.
HIL_GPS message: http://mavlink.org/messages/common#HIL_GPS
GPS_INPUT message: http://mavlink.org/messages/common#GPS_INPUT

khancyr, 2016-12-07:
@rmackay9 or we can make a mavros plugin for GPS_INPUT. I could do it!

closed 2016-12-08 by rmackay9

rmackay9, 2016-12-08:
@khancyr, that would be great!  that's definitely a better solution.  @291 also said he might do it... but whomever is faster wins! :-)

291, 2016-12-09:
@khancyr, are your plugin completely work ? Now, I have problem my plugin could not send GPS_INPUT stably. I'm testing my plugin but I can't found it solution.

khancyr, 2016-12-09:
@291 I haven't start working on that yet and the best I can do I write the plugin I don't have the material to test it

291, 2016-12-10:
@khancyr I found why I couldn't send GPS_INPUT stably. That's might my hardware problem(Erle Brain2).
When I use "ros::Time::now()", I get wrong time. I tested my plugin without using "ros::Time::now()", I could send GPS_INPUT stably.