ArduPilot/ardupilot #5356: All: Dataflash log file size rotation

gmorph, 2016-12-09:
Some users run their vehicles for a VERY long time.  In the case of boat users they often run 24 hours a day over several days.  At the moment his generates one monolithic huge log file.  First question is what happens when the SDCard becomes full and its only got 1 file on it and its the file currently being written to?  Ideally at a certain point we would split the log file into separate files of say 1Gig each for ease of handling.  There are a few hurdles to overcome.  
All the self describing information, parameters etc are gathered at the beginning of the log file so we either have to copy that stuff to the new log file as its created and treat each log file as self contained OR assume that subsequent log files are part of the first one and we need to stitch them back together after download or have the log file processing tools be able to handle multiple files.
If we do split into multiple files we still need to decide what happens when the SDCard becomes full.

lvale, 2016-12-10:
Easy, A Parameter to enable rollover logs, ie, when the card is full erase the oldest log and start a new log file. If the Parameter is to not enable rollover logs, than logging should stop. User choice. Shouldn't also the log files be less than 4GB due to file system constraints ?

magicrub, 2016-12-10:
Luis,

That is already in place on boot. Problem is if there are zero logs and
then you create a single huge log then it will sort of just fail right now.
Grant is talking about logging for multiple days, we need a way to stop
logging, then start a new file with new headers and such and continue
logging and then if you run out of room then we do what you were saying.

lvale, 2016-12-10:
Yes @magicrub I have a 4GB full card of logs. This situation touchs everybody. My 2 suggestions are to enable the rollover and limit the file size to 4GB for a single log file.

magicrub, 2016-12-10:
File size should be smaller. I agree with the 1GB limit that someone mentioned earlier

lvale, 2016-12-10:
@magicrub the 4GB is absolute max due to file system restrictions, and it's not a good size (too large) to deal. The 1GB is a good proposal

peterbarker, 2016-12-11:
On Thu, 8 Dec 2016, Grant Morphett wrote:

> Some users run their vehicles for a VERY long time. In the case of boat
> users they often run 24 hours a day over several days. At the moment his
> generates one monolithic huge log file. First [...]

Logging will stop within a few percent of filling the SD card to avoid 
corruption which NuttX does not handle well.

> log file into separate files of say 1Gig each for ease of handling. There
> are a few hurdles to overcome.

Rolling the log file over at 1G seems reasonable to me.  We can 
parameterize it the first time someone complains their 
Linux-based-autopilot has 1 1TB SSD on it and that this is dumb :-)

> All the self describing information, parameters etc are gathered at the
> beginning of the log file so we either have to copy that stuff to the new
> log file as its created and treat each log file as [...]

We could add a "critical" message to mark a log as a "continuation log", 
possible referencing the original log somehow.  Maybe we could have a META 
message which has a UUID in it, and continuation-logs reference that?

> If we do split into multiple files we still need to decide what happens when
> the SDCard becomes full.

Yes, this is the thorny part.  Does the user want to keep the most recent 
logs or the oldest logs?  Sadly, a parameter is probably the only option 
here.

magicrub, 2016-12-12:
I reluctantly agree to all if that, including the params.

patrickelectric, 2024-07-05:
At the moment we have 6.5GB of tlog files, and a single tlog of 2.1GBs, it would be nice to have a log rotation and configuration in ardupilot!

[image]