ArduPilot/ardupilot #5357: Adding support for cheap ultrasonic range finder

katzz0, 2016-12-09:
#### Issue details
Please add support for the I2C cheap Range Finder which can be assembled by using cheap SR04 ultrasonic sensor and attiny. Firmare is located here https://gist.github.com/ajfisher/1d57c5f845c376f04fbb

#### Version
Latest

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Doesn't matter

#### Hardware type
Pixhawk as minimum

#### Logs
No logs

squilter, 2016-12-10:
We don't support the one built into PX4Flow because it's unreliable. This will presumably be even less reliable.

rmackay9, 2016-12-10:
We can actually already support a sensor that sends it's distances using the DISTANCE_SENSOR message.  So if the attiny could send these mavlink message then it would likely just work.

tomarigr, 2016-12-10:
im sorry if i am a bit behind, but i am confused. Doesn't this work anymore?
http://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html

katzz0, 2016-12-10:
The solution by using MAVLink protocol looks universal but if I correctly understood the packeges have to be sent via I2C? I also see drivers for I2C sensors. If the solution with MAVLink won't be implementable it seems easy to just copy driver for some I2C sensor and set up parameters for the specified one the problem only will be binding with other parts of the system.

And a few words about analog solution. This is possible but the range of DACs on attiny is 8 bits and it isn't enough to send precision information for 400 cm it will be 1.6 cm per bit. I2C is much reliable solution.

rmackay9, 2016-12-10:
@tomarigr, the maxbotix sonar are still supported but that's a different type of sonar compared to the type that @katzz0 has requested.  The request is for this $4 sonar: https://www.sparkfun.com/products/13959.  It's a relatively short-range sensor (4m) and normally the effective range is about 60% of the advertised range.

tomarigr, 2016-12-10:
thanks @rmackay9, i wrote a small arduino code, using a MCP4725 dac and convert the SR04 output to 0~3.3v. 
Does arducopter handle more the measurements (ekf etc) if this type of sonar is selected? (analog)
why should we use the mavlink way for the sonar? what are the advantages featurewise?

lvale, 2016-12-10:
@tomarigr If your solution also requires some external components it would be great if you could prepare a page for the wiki documenting the external hardware/firmware changes to use the SR04.

Although they aren't the greatest sonars they are very cheap and most DIY users might have a few on their toolbox so getting the SR04 sending an analog signal (which is supported on the latest copter version) won't need any changes to the main codebase.

tomarigr, 2016-12-10:
@everyone :) this is what i am using:
- Arduino nano and MCP4725  DAC (easy, fast, reliable!)
- good quality 5v ubec
- DS18B20 temp sensor
inspired from this project by Alexander Buckley: 
https://bitbucket.org/albuckley/alx_hc-sr04_pixhawk/wiki/Home
i connect signal and ground to pixhawk like this:
http://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html

the uploaded scetch is used to drive the SR04 and DAC 
code is lame please be patient!!
im sure u can do your magic with it and make it even more roburst!
then we can make the wiki :)

[PiXSonar_2.zip](https://github.com/ArduPilot/ardupilot/files/643809/PiXSonar_2.zip)
[image]

mirkix, 2016-12-30:
The [BBBmini](https://github.com/mirkix/BBBMINI) ArduPilot Cape support the HC-SR04. But the performance (in flight) is very bad.

tomarigr, 2016-12-31:
an other cheap rangefinder is VL53L0X (2$)  it can connect at the same arduino nano with hc-sr04  and assist for under 2m distance.
Its a mini laser finder but only 2m

katzz0, 2017-01-12:
I tested HC-SR04 with arduino nano backpack. Digispark with attiny 85 didn't work due to start up delay and I think ardupilot doesn't  recive any messages from the sensor at start up time and thinks that there is absence of the sensor. I selected maxbotix sonar I2C and changed the code for arduino nano in accordance maxbotix specification, it used 70h address and so on. But results of the test have disappointment me because quadcopter always tries go up or go down and it didn't want to hover on the same height. I haven't checked logs yet but I hope I'll find the reason why it behaves itself like this. It strange so I tested sensor alone and on quacoper in hand and it showed good results with dinstance detection on screen.

So I can say that my copter flights better and hovers without sonar.

tomarigr, 2017-01-12:
hello this weekend i will give it a try again with the solution 4 replies above and provide full logs and video to discuss.

Saxin, 2017-03-12:
@katzz0 Can you you please share the code or link to repo, I'm trying to do something similar more or less ?
Thanks.

rmackay9, 2017-03-13:
Since the last time we discuss, Copter now supports object avoidance using 1-dimensional sonar/lidar pointing forward, upwards, sideways, etc.  We still only support 2 lidar maximum but it's not too difficult to add support for up to 4.  The difficulty becomes finding ports to attach all the sonar/lidar.
http://discuss.ardupilot.org/t/copter-object-avoidance-with-upward-facing-lidar

proficnc, 2017-03-13:
If we use i2c can we have more? We can address shift them..

lvale, 2017-03-13:
most efficient way would be to have a cluster of prox. sensors attached to a simple processor (Arduino like) that would send the direction and distance of each sensor in sequence on either I2C (harder due to cable length) or other (UAVCAN?) or serial: Here's an idea for a product @proficnc  :)

patrickpoirier51, 2017-03-31:
@lvale  Yep, that something taht we are workin on here : https://github.com/ArduPilot/ardupilot/issues/5605
[image]

devanshg25, 2017-04-09:
Hi all,
Im trying to do obstacle avoidance with 9 Maxbotix LV EZ0 sensors on a quadcopter such that 8 are horizontal to cover 360 degrees, and 1 below for altitude hold. Im measuring distances from these on an arduino and then sending DISTANCE_SENSOR Mavlink messages to Pixhawk from it with the correct ORIENTATION value corresponding to the measurement being sent. However, on the PIxhawk, I can only find parameters max. for 4 rangefinders. 

**Is there any way I can get all my sensors to work with the Pixhawk or is 4 the maximum number of rangefinders that can be supported?** And if 4 is the max, then is there any other way I could do obstacle avoidance on the Pixhawk with these 9 sensors? @rmackay9, could you possibly help on this one?

rmackay9, 2017-04-10:
@devanshg25, I replied to your email separately but in any case, for the horizontal range finders, there's no need to setup the RNGFND_ parameters.  Instead set PRX_TYPE to "MAV" and that will consume all the horiztonal DISTANCE_SENSOR messages.

devanshg25, 2017-04-10:
Thanks a lot for the quick reply @rmackay9, that makes a lot of sense! I had a couple of other queries and would really appreciate your help on them:

1) Is there any thing I need to do to ACTIVATE obstacle avoidance? I take off in Stabilize mode in which I believe there is no obstacle avoidance, but then switch to AltHold or Loiter - so should obstacle avoidance start automatically when I switch into the new mode?
2) How can I set the distance to be maintained from obstacles? I just found the AVOID_DIST_MAX parameter which seems to be for this purpose, but it looks like its only for non-GPS modes and has a min distance of 3m. Is there any way I can change the distance to be kept to maybe 1m in nonGPS/GPS modes?
3) Is there any way to test the obstacle avoidance to see if obstacles are being detected without the risk of crashing into those obstacles? The Mission Planner [display](http://ardupilot.org/dev/_images/code-overview-object-avoidance4.png) shown in the [wiki](http://ardupilot.org/dev/docs/code-overview-object-avoidance.html) would really help but I have no idea how to actually get that display working in Mission Planner?

Thanks a lot in advance!

rmackay9, 2017-04-10:
1. there are some AVOID_ parameters but it should be on by default.  There's also an auxiliary switch option (ch7, ch8, etc) which can be used to turn avoidance on and off.
2. The distance is currently hard-coded to 2m for GPS modes.  Feel free to raise a PR though to add in a new parameter if you're up for that.  It's something we should add.
3. the Mission Planner has a proximity viewer screen that pops up and shows the distances in the 8 horizontal directions.

No troubles, best of luck.

anuchitd, 2017-04-15:
Hi all,
I modify code from https://gist.github.com/ajfisher/1d57c5f845c376f04fbb
** Arduino Nano.
1 ) hcsr04_backpack.ino
   ISR(PCINT0_vect) {....
   //  duration = micros() - ping_emit_time;
  duration = ((micros() - ping_emit_time)/100)*1.785;   //+-2 cm.
  //
2 ) includes.h
  #define DEFAULT_I2C_SENSOR_ADDRESS  0x70 //Maxbotix Ultrasonic Sensors

Firmware APM:Copter V3.5-rc4
Test i2c port with 
1 ) Pixhawk work.
2 ) Navio+ work.
3 ) Navio2 work.
 Set Parameter
RNGFND_GNDCLEAR  12
RNGFND_MAX_CM	   400
RNGFND_MIN_CM	   2
RNGFND_TYPE	           2

anuchitd, 2017-04-15:
code..
[hcsr04_backpack.zip](https://github.com/ArduPilot/ardupilot/files/923670/hcsr04_backpack.zip)

inligo, 2018-06-12:
Hi all 
I need your help in the following simple(?) issue: I have connected through I2C two ARDUINOs (master a MEGA and slave a NANO). Using the following code they communicate correctly:
**For the master:**
#include <Wire.h>
String msg = String("");
void setup() 
{
  Wire.begin();                // join i2c bus 
  Serial.begin(9600);       // start serial for output
}
void loop() 
{
  Wire.requestFrom(10, 27);  // request n=27 bytes from slave device #10
    while (Wire.available()) 
    {                                     
      char c = Wire.read();  // receive a byte as character
        
      Serial.print(c);            // print the character
    }
delay(1000);
Wire.end();
}
And for the slave:
#include <Wire.h>
void setup() 
{
  Wire.begin(10);                              // join i2c bus with address #10
  Wire.onRequest(requestEvent);     // register event
}
void loop() 
{
  delay(1000);
}
                          // function that executes whenever data is requested by master    
void requestEvent()       // this function is registered as an event
{
  Wire.write("41.139234,24.912345,67.25; "); // respond with message of n bytes                     
}
I must replace the master ARDUINO MEGA with the PIXHAWK flight controller of a hexa. What is the code in python scripting, that will replace the above cpp code, in order to communicate through I2C with the slave ARDUINO NANO and read the above sentence "41.139234,24.912345,67.25;”? I use python scripting through Mission Planner to control the hexa.
Thanking you in advance.

anish-pratheepkumar, 2019-11-13:
> 1. there are some AVOID_ parameters but it should be on by default.  There's also an auxiliary switch option (ch7, ch8, etc) which can be used to turn avoidance on and off.
> 2. The distance is [...]

Hai is there any update on the minimum distance of obstacle avoidance, is it still 3m in non-gps mode?
So is it like if AVOID_ENABLE is activated and if any obstacle is present in 3m range of drone, it will not move in that direction? And no middle element like arduino is required for obstacle avoidance, only Pixhawk FC and a single range finder is enough, is this understanding correct?
Also can you please give any latest information regarding the integration of cheap ultrasonic range finder to the copter using pixhawk FC, for obstacle avoidance ? I found some discussion above but could not understand what  DISTANCE_SENSOR message from range finder means, is it a parameter in MissionPlanner?
Any help will be appreciated!

peterbarker, 2021-07-30:
As it turns out, we added native support for this sensor.

closed 2021-07-30 by peterbarker