ArduPilot/ardupilot #5359: Accelerometer calibration sluggish and failing to complete folling update to Arducopter 3.3.4

GTMonster, 2016-12-09:
#### Issue details
With version 3.3.4. I can't complete the accelerometer calibration on my Pixhawk 2.4.6. 

The process goes so sluggishly between each of the accelerometer calibration steps. 
For example - Place place flat - click continue - wait 2 minutes - screen finally displays move onto left side - click continue - wait 2 minutes - screen finally displays move onto right side - and so on however before completion, the accelerometer calibration fails.
Reverted to 3.3.3 accelerometer calibration all working as normal again and able to complete setup.

#### Version
3.3.4

#### Platform
Copter

#### Airframe type
X4

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2016-12-09:
First of all, this is a support question and those should be posted at the forum (http://discuss.ardupilot.org). Then I'm not sure why you are using 3.3.4 in a Pixhawk as that was a specific version for Pixracer. If you meant 3.4.3, then post in the forum, together with a log.

closed 2016-12-09 by OXINARF

GTMonster, 2016-12-09:
My apologies. Ive never used this before.