ArduPilot/ardupilot #5367: Plane: Cruise Mode Throttle/Pitch Control Bug

SlipstreamFPV, 2016-12-11:
#### Issue details

In CRUISE mode, the throttle and pitch control can get into bad oscillations. In other auto modes like RTL, LOITER and AUTO this doesn't happen. Different throttle values can be used to nudge the cruise airspeed which will sometimes bring it out of the oscillations. The values that oscillate and the values that work are random and vary constantly.

Discussion:
http://discuss.ardupilot.org/t/throttle-and-pitch-auto-control-issues/12081

#### Version
I've encountered this problem with every version from 3.4 up to 3.71

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
My Twin Dream

Tridge may have been able to repeat the issue with Cruise mode on his Parrot Disco as well.

#### Hardware type
Pixhawk
NEO-6M GPS + MAG
uBlox CN-06 GPS
Digital (Also tried analogue airspeed sensor)
Mauch current sensor
Turnigy 50090MG Servos  (Also tried HiTec HS-65MG)

#### Logs

You can see the issue on pitch and throttle just before 2min, 4min and 9min:
https://www.dropbox.com/s/vtksvwl4jstnvca/11.BIN?dl=0

You can see the issue here from 8min through to 16min:
https://www.dropbox.com/s/l76sh5y2m33st9t/12.BIN?dl=0

You can see the issue here just after 41min, just before 44min and 45min:
https://www.dropbox.com/s/6ictr9d62wh9tvo/10.BIN?dl=0

On the final log (10.BIN) RTL is engaged right after using cruise to show that the throttle and pitch work properly in RTL mode.

SlipstreamFPV, 2017-01-05:
Just to update, I've replaced the pixhawk with a new one and the issue still exists. Thought it was best to try a new one to eliminate the chance of this being caused by a faulty pixhawk.

magicrub, 2017-01-05:
This is still a mystery for us, sorry. Meanwhile, are you able to fly on a
different mode to avoid the behavior?

SlipstreamFPV, 2017-01-06:
Definitely flying other modes instead at the moment. Cruise mode is just so awesome when its working well. Much better than what most other plane OSDs / autopilots offer.

SlipstreamFPV, 2017-02-27:
@magicrub @tridge 

Would be great if some action was taken on this. I'm experiencing the same issue on a new build as well.

magicrub, 2017-02-28:
@tridge can you take a look?

SlipstreamFPV, 2017-03-09:
Any new insights after the dev call? I tried to get on but had to go to work.

OXINARF, 2017-03-09:
@SlipstreamFPV It was mentioned but a full log analysis is needed which is out of the scope of the dev call. @tridge said he would try to look at it, but his time is very limited so no timeline provided. I'll try to bug him now and then to sort this out.

SlipstreamFPV, 2017-04-10:
@OXINARF @tridge @magicrub Who do I have to bribe to get this looked into? 😄  Seriously! Arduplane is so good on all of my planes apart from this one bug on this one airframe. And the airframe flys great in Stab, FBWA, RTL etc. Could it be caused by the different auto throttle handling in Cruise and FBWB vs every other auto mode?

magicrub, 2017-04-10:
I think if we knew we would have fixed it by now. Sorry! This is definitely a gremlin that we are not happy about. You're the only one we've heard of complaining about it so I wonder if there's something about your config that we've missed?

SlipstreamFPV, 2017-04-10:
Thanks for writing back. I don't want to be a pain about it, I wish I could find the setup issue if it is that :) I've tried so many different tuning values and tried defaults and auto tune (which works great on my other planes). The only bit of hardware I haven't changed out at this stage are the two GPS units. I've swapped them around so the primary is different though and turned off auto switching so it probably rules them out too. 

I'm happy to pay someone to look into my logs if anyone has time!

EShamaev, 2017-04-11:
Can you please share the value of the RCx_DZ settings you have setup on your autopilot?

EShamaev, 2017-04-11:
Also noticed, that even in steady level your RC1 and RC4 are out (trim+/-dz) values. Is there a reason for that?

EShamaev, 2017-04-11:
And this problem is present in other flight modes as well, not only CRUISE. It is not so obvious and noticeable, but it can be observed in Loiter and RTL. It shows as a sudden continuous glitches in _TAS_rate_dem

SlipstreamFPV, 2017-04-11:
@EShamaev Thanks so much for taking some time to look at the log. I have 30 DZ on each channel but Pitch which only has 10 DZ. Even though the input from the Rx seems to be pretty solid only going up and down by 1 I will try DZ 30 on pitch as well if you think it might help.

For a short while I accidentally had rudder and ailerons trimmed counter acting each other - when I graph the outputs now the mean of each after a long flight is +/- 5 on the trim value.

How do I graph _TAS_rate_dem, I can't find it in the list? What would cause spikes in that? Is the raw airspeed data too rough? I've tried both analogue and digital airspeed sensors in the plane.

EShamaev, 2017-04-11:
In your logs, RC1 input and RC4 inputs are diffidently out of trim. They are 35-40 points away from trim, which is more than DZ.
_TAS_rate_dem is dspdem in TECS.
For now I have no explanation on what causes it, but it was found in other flight modes, not only in cruise. However in cruise it is very repeatable.

SlipstreamFPV, 2017-04-11:
@EShamaev Thanks again for your quick response.
[image]
[image]

Is this what you are referring to? They seem to be in line with each other now although in the past with those older logs they have been out. The issue is still present regardless.

You're very right regarding TECS.dsp_dem. It has spikes throughout. On my other planes this value sits at 0 for the whole flight. (One of them is even running the original pixhawk out of this plane before I changed it to see if that was the problem).

EShamaev, 2017-04-11:
No, I am referring to trim values of these inputs.
[image]
[image]

You have a value of 1585 as trim but in params it is set as 1550 with DZ of 30. 1585 > (1550+30). So cruise controller is never placed in cruise mode.

EShamaev, 2017-04-11:
As for TECS.dsp_dem, glitches are exactly where you described strange behavior.

[image]

But as you can see they are also present in other flight modes:

[image]

SlipstreamFPV, 2017-04-11:
@EShamaev  Thanks again. Yes - I've fixed the RC trim values. I reset all of the parameters on many occasions when I made those logs so they might not have been trimmed properly. They definitely are now. The images I posted from the new logs you can see that the input mean is almost exactly the same as the trim value.

You are correct - I haven't noticed it in other flight modes. Good observation. 

@magicrub Sorry to bother you again, Could you shed some light on what factors influence _TAS_rate_dem?

SlipstreamFPV, 2017-04-20:
Someone out there must have an idea of what would cause _TAS_rate_dem (TECS.dspdem) to increase in value as opposed to staying at 0 like every other plane I have or have seen?!

I'm lost on this one and about to give up on this plane entirely since I've changed all the hardware apart from the airframe itself now.

23suleyman, 2022-12-02:
Hello there
i have the same problem with cube black
There is no problem in fbwa, but in auto rtl and cruise mode, the gas is constantly on and off and flight becomes impossible.
Do you have any suggestions for a problem-free solution with the same plane matek f405 wing installed on my plane with the latest version fw?