ArduPilot/ardupilot #5369: ARMING_REQUIRE=2 and auxiliary throttle channels

mothacehe, 2016-12-12:
#### Issue details
Hi all,

When setting ARMING_REQUIRE to 2, no pwm is sent to throttle channel (3 by default) while disarmed. When adding an auxiliary throttle channel (e.g RC5_FUNCTION=70), a min pwm is always sent even with ARMING_REQUIRE=2 on this channel.

To prevent accidental auxiliary motor launch, wouldn't it be a better idea for auxiliary throttle channel 
to behave like the main throttle channel ?

Thanks,

Mathieu

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Disco

#### Hardware type
Disco

WickedShell, 2017-05-29:
@tridge Is this fixed with the SRVChan stuff?

tridge, 2017-05-29:
I haven't specifically tested this case, but the SRV_Channels change should mean it is consistent. @mothacehe, can you please test the latest master?

IamPete1, 2021-08-16:
Fixed by https://github.com/ArduPilot/ardupilot/pull/18050 we now treat throttle left and right the same as main throttle and they all honor the safety mask.

closed 2021-08-16 by IamPete1