ArduPilot/ardupilot #5370: Traditional Helicopter: 4 swash servos

0n3-70uch, 2016-12-12:
#### Issue details
Hello, I want to add a 4th swash servo to the arducopter code. I changed the code in "AP_MotorsHeli_Single.cpp" and "AP_MotorsHeli_Single.h". Here you can find the code with a 4th swash servo:
https://drive.google.com/drive/folders/0B_Ez_duqb1SDVmoyZ0JGeTlsLWs?usp=sharing

I want to compile this code but I receive this error:
`/home/fabian/Downloads/ardupilot/libraries/AP_Motors/AP_MotorsHeli_Single.cpp: In member function 'virtual void AP_MotorsHeli_Single::calculate_roll_pitch_collective_factors()':
/home/fabian/Downloads/ardupilot/libraries/AP_Motors/AP_MotorsHeli_Single.cpp:332:25: error: array subscript is above array bounds [-Werror=array-bounds]
         _rollFactor[CH_4] = cosf(radians(_servo4_pos + 90 - _phase_angle));`

Greetings Fabi

#### Version
3.4.3

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Traditional helicopter with four swash servos

#### Hardware type
pixhawk (px4-v2)

#### Logs
_Please provide a link to any relevant logs that show the issue_

fhedberg, 2016-12-12:
https://github.com/ArduPilot/ardupilot/pull/3245

0n3-70uch, 2016-12-12:
Nice to see that there are some other guys with the same idea.

OXINARF, 2016-12-12:
This is a development question and those should be posted at the forum (http://discuss.ardupilot.org) or asked about in Gitter (http://gitter.im/ArduPilot/ardupilot).

Look at the PR @fhedberg posted for a correct implementation.

closed 2016-12-12 by OXINARF

0n3-70uch, 2016-12-13:
Ok thank you.