ArduPilot/ardupilot #5373: Sim_vehicle.py: gdb-stopped is not implemented

khancyr, 2016-12-13:
#### Issue details
gdb-stopped is not implemented
https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L736
#### Version
Master : at https://github.com/ArduPilot/ardupilot/commit/76106889d59706abd44bf9ac4a2a250a96008e41
#### Platform

peterbarker, 2016-12-14:
Patches are required to both pymavlink and MAVProxy to get this happening - otherwise MAVProxy attempts to connect to the simulated vehicle via tcp, gets a "connection refused" and bombs out.

https://github.com/ArduPilot/MAVProxy/pull/352
https://github.com/ArduPilot/pymavlink/pull/39

If someone can independently test those patches I think I can push them in.

khancyr, 2016-12-14:
@peterbarker Ok I add this on my pile of Todo !

peterbarker, 2017-12-19:
@khancyr I'm guessing your pile isn't getting any shorter ;-)

khancyr, 2017-12-19:
nope I have LIFO pile ! But I had made some comments and test on both linked PR !

peterbarker, 2024-07-31:
This works now.

closed 2024-07-31 by peterbarker