ArduPilot/ardupilot #5375: Feature request - protection against RC hang

night-ghost, 2016-12-13:
sometimes RC receiver stops working in unexpected ways, example: https://www.youtube.com/watch?v=HGblaQfv7QQ
On this video powerfull video TX caused RC hang.

In code I found only 2 failsafe checks - no packet for a some time and throttle below limit (except stop of motors  in the case of loop hangup which for some reason don't cause chute release - Copter::failsafe_check() ).

I think that there should be 3rd check - some RC channel value should be changed in last 10 seconds, to prevent flyaway  in case of receiver's malfunction 

#### Version
git

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover

Naterater, 2019-01-04:
I think this should probably be addressed by other issues covering bad RC inputs. Such as https://github.com/ArduPilot/ardupilot/issues/10139

peterbarker, 2021-07-30:
I was told quite clearly that static RC inputs for 10 seconds are entirely expected, so this is anon-starter, I'm afraid.

closed 2021-07-30 by peterbarker