ArduPilot/ardupilot #5376: Copter necessarily will crash in case if device on the I2C bus hangs.

night-ghost, 2016-12-13:
Because main loop will stop working by bus_i2c_reset() which will cause stop of motors  in Copter::failsafe_check(). It is clear that we here can do little, but at least if the parachute present code should to release it.

#### Version
git

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

OXINARF, 2016-12-13:
Can you explain better what you are requesting? Is it that you want parachute released when the failsafe check is activated?

night-ghost, 2016-12-13:
At least yes - because we CAN to have parachute system but not release it in case of problems. So 
1) throttle should be not in minimum but rather THR_MID or slightly less
2) just after disarm parachute should be released (may be with standard check about altitude and position that were remembered).  Errorneous parachute release is much less problem than the copter with all  equipment fallen from the sky.

IamPete1, 2021-07-25:
`bus_i2c_reset` is gone, I'm not %100 sure what the relation to parachute is, I'm going to close, feel free to re-open if there is still issue.

closed 2021-07-25 by IamPete1