ArduPilot/ardupilot #5384: Wait the end of DO commands before finish mission

khancyr, 2016-12-14:
#### Issue details
When doing an auto mission, the mission stop when the last nav command is reach even if DO command are still running. (https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mission/AP_Mission.cpp#L210)
The better logic should be to wait the end of the DO command if it already running,or finish the list of Do command in the mission

#### Version
Master

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

magicrub, 2016-12-14:
Can you provide a use-case example?

rmackay9, 2016-12-14:
I can imagine some use-cases. for example putting a do-gripper command right at the end so that it drops a package after landing.  Or a do-mount command to move the camera to a retracted state.
It should be possible to work around this by adding a nav-delay as the last command but fixing it so that the mission doesn't complete also makes sense.

khancyr, 2016-12-15:
Yep as Randy said one use I got is to free rover from copter after landing .
But the other I got now is the implementation of on spot turn for rover using do_yaw for rover, having the possiblity to turn it at the end of a mission is a great plus for something like defining a ROI.

The modification shouldn't be hard to do, so do I need to open a post on discuss for that or try to come on a dev call for that ?

rmackay9, 2016-12-15:
@khancyr, anywhere you can grab people really?  so Gitter or Mumble during asia hours.  It'll certainly be changes within the AP_Mission library..  I'm not sure exactly how it should work but I guess it's either:
1. the last nav command needs to keep running until the last do-command completes.  I'm not sure this will work very well.
2. the last nav command completes but the mission does not until the last do-command completes.
During testing, the plane code is what I would worry about most.  Even if the last nav command's verify stops being called, Copter will just sit there.  I'm less sure what plane will do if it's final nav command's verify isn't called continuously.