ArduPilot/ardupilot #5396: Helicopter: Bug in Acro mode

0n3-70uch, 2016-12-16:
#### Issue details
The swash is not levelling correct. And the swash make some random moves. Happens only in Acro flight mode. A video show more than thound of words:
https://youtu.be/Mlb3pnwZgL8

#### Version
3.4.3

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Traditional Helicopter

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

R-Lefebvre, 2016-12-16:
That's interesting.  I haven't seen anything like that before.  Can you provide your parameter file?

0n3-70uch, 2016-12-16:
Here is the parameter file.
Same problem after a parameter reset (only acc, compass and rc calibration).

[Helicopter.txt](https://github.com/ArduPilot/ardupilot/files/657343/Helicopter.txt)

0n3-70uch, 2016-12-21:
Hello,

I want to ask if there is another person with the same issue...

Greetings Fabi

0n3-70uch, 2017-05-11:
Hello,

There is a long topic about this problem:
http://discuss.ardupilot.org/t/unusual-swashplate-behavior-using-ac-3-4-5-in-acro-with-full-cyclic-input/

Greetings
Fabian

tridge, 2017-09-20:
regarding the "random moves", that is caused by the target attitude in the acro controller wrapping. I completely agree that it is confusing and rather disconcerting for a user doing ground testing. It won't actually impact the flight of the vehicle, but we should fix it anyway.
I had a discussion about this with @lthall and we discussed adding a angle limit to the difference between target and current attitude to control_acro.cpp. That would avoid the ground test issue (as we could no longer wrap the target). Leonard has an alternative approach he is looking at for fixing this that he'd like to try first.

0n3-70uch, 2017-09-20:
Hi, nice that you are working on this topic... thanks.

ChristopherOlson, 2017-09-27:
@tridge can this issue be closed? It doesn't do it anymore with Copter 3.5.3. Although I can't find it, it seems I saw a PR from @rmackay9 some time back that limits the ghost error so it doesn't continually "wrap" if the actual aircraft attitude doesn't change.

maxdrougge, 2017-09-28:
Hello. Is this issue closed? Is software 3.5.3 safe to fly with? If I'm not mistaken this version supports 4 servo swash plate?

ChristopherOlson, 2017-09-28:
No, the issue is still open as the described swashplate behavior still happens in 3.5 with FBL helicopter. It does not happen with flybar.

0n3-70uch, 2017-09-28:
Hi Christopher, I updated my first post to "version 3.4 - 3.5.3".

bnsgeyer, 2017-10-02:
@lthall I'm interested in learning what your plan forward with the swashplate behaviors will be for unarmed and armed states while on the ground.  Please fill us in when you get the chance.  I would propose a fix for the unarmed state in a pull request but I'm interested in what you've got brewing.

ChristopherOlson, 2017-10-28:
This was fixed by PR #7136

closed 2017-10-28 by ChristopherOlson

0n3-70uch, 2017-10-28:
Very nice :)