ArduPilot/ardupilot #5419: Bug report: Impact detect logic seems wrong

nocternal, 2016-12-20:
#### Issue details
In is_flying.cpp
the code to judge impact seems wrong. I think it should use accel.z compare with threshold which can represent for load factor,rather than use accel.x, which represents for horizon acceleration.  When impaact to ground the z axis acceleration will change fast and big, not horizon x axis. @magicrub  

the code:
```
...
else if (ins.get_accel_peak_hold_neg_x() < -(g.crash_accel_threshold)) {
                // large deceleration detected, lets lower confidence VERY quickly
                crash_state.impact_detected = true;
...
```

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

tridge, 2016-12-20:
Z axis doesn't necessarily change rapidly when the plane hits the ground. If the plane hits nose in then it is primarily X that changes. If it hits side on then Y changes. 
What always happens though is a large, rapid change in fwd speed, which is what X accel detects.
I wasn't involved in the development of this code, but I believe Tom did quite a bit of testing before settling on the method he used.

closed 2016-12-20 by tridge

nocternal, 2016-12-21:
I agree that in some condition X or Y axis changes bigger than Z axis. But most of times for a normal landing I still believe Z axis's change is biggest. For example, here is a figure of one of my plane normal landing(electric Plane:skywalker X8),
[image]
Only with friction of ground the Acc X only change a little ,what's more, I found in Parameters.cpp
,here is the definenation of CRASH_ACC_THRESH:

    // @Param: CRASH_ACC_THRESH
    // @DisplayName: Crash Deceleration Threshold
    // @Description: X-Axis deceleration threshold to notify the crash detector that there was a possible impact which helps disarm the motor quickly after a crash. This value should be much higher than normal negative x-axis forces during normal flight, check flight log files to determine the average IMU.x values for your aircraft and motor type. Higher value means less sensative (triggers on higher impact). For electric planes that don't vibrate much during fight a value of 25 is good (that's about 2.5G). For petrol/nitro planes you'll want a higher value. Set to 0 to disable the collision detector.
    // @Units: m/s/s
    // @Values: 10 127
    // @User: Advanced
    GSCALAR(crash_accel_threshold,          "CRASH_ACC_THRESH",   0),

Look, it says it can has a value more than 25 m/s2, Without a brake system, it's not a common condition can get X axis has a 25m/s2(may be use reverse thrust can archive)

In a word, maybe it's not wrong to use AccX, but first we should use AccZ to detect load factor change when impact happen  @tridge   @magicrub

OXINARF, 2016-12-21:
@nocternal The part you are talking about is about a crash, not a normal landing.

nocternal, 2016-12-21:
In fact if the normal landing is designed at a low speed, and plane has a big |Cmα| means it has good stability. Then the delta elevator degree maybe big, In my plane it can use half or more of my elevator, if there is no impact detect for this kind of landing, the big up elevator can cause plane's pitch become bigger again after hit on the ground and cause bounced landing.
So even in normal auto landing, we may need detect hit on the ground and make elevator pwm to 1500