ArduPilot/ardupilot #5420: Potential issue in AlexMos driver

SalahuddinAsh, 2016-12-20:
#### Issue details
Hi everyone, 
I'm trying to control an AlexMos gimbal controller using ODroid (ODroid -> Pixhawk -> AlexMos)
I've sent the right MAVLink message from ODroid to Pixhawk and monitored the output of Pixhawk to AlexMos by connecting this output to the PC. I found that the output is not correct. 
Pixhawk just sends empty messages to AlexMos. Something like this
`3e  43   0d   50   02   f5   00   00   00   f5   00   00   00   f5   00   00   00   e1`
You can notice that the comand id is correct which is "CMD_CONTROL = 'c' = 43"  and the rest of the fields of the message are correct but empty anlges!!
I followed the flow of control in the master branch and found it as following:

#
**command_logic.cpp** 
start_command(cmd) → do_mount_control(cmd) 
			|
			|
		AP_Mount_Backend.cpp 
		set_angle_targets(roll, pitch, yaw) // sets the target angles 
		
#
**GCS_MAVLink.cpp**
handleMessage() 
	|
	|
**AP_Mount_Backend.cpp** 
control(pitch, roll, yaw, mount_mode)    →     set_angle_targets(roll, pitch, yaw) // sets the target angles

#
**ArduCopter.cpp**
update_mount()
	|
	|
**AP_Mount_Alexmos.cpp**
          update()	  → 	control_axis(angles, in_degrees) 	→ 	send_command(cmd, size)

I found that **update()** should call **control_axis()** which should call **send_command()** to really send the command. This doesn’t happen in case of MAVLINK_TARGETTING mode. The target values are set but the send_command is not called.

Could someone please tell me if I'm right?
Many thanks

OXINARF, 2016-12-20:
There has been talks about this before. I believe you are right but I don't have any Alexmos to test it and the comment there always refrained me from changing things. There is a PR opened to add the control_axis (#4420) and there is also a PR opened with many improvements to Alexmos (#5297).
I guess this is just an issue with contributed (and unmaintained) code. If you can test the commit in #4420, I can bring it in.

Regarding the DO_MOUNT_CONFIGURE you are wrong, as that would be handled by the backend. It was a decision by Randy to not support that command because (https://github.com/ArduPilot/ardupilot/pull/2696#issuecomment-138048761):

> I couldn't think of why someone would use this instead of using he DO_MOUNT_CONTROL message which also allows setting the mode. The only other thing that do-mount-configure offered was changing which axis are stabilized which I don't think is actually a useful feature because these are just 2 of about 20 parameters that need to be set to configure a gimbal.

SalahuddinAsh, 2016-12-20:
@OXINARF Thanks a lot for your reply. I didn't know that someone talked about it before. 
There something that I can't understand clearly. If we are agree that the **send_command()** function is not called in **MAVLINK_TARGETTING** mode, then how the Pixhawk sends the **CMD_CONTROL** message with angles values = 0s? It shouldn't send any, right?
It doesn't send any messages until ODroid sends it the DO_MOUNT_CONTROL message, then Pixhawk starts to send CMD_CONTROL messages!!
I think to test this commit I can just add this additional line of code in my code version, build it, download it on Pixhawk, right?
I'll do it as soon as possible and let you know.
Many thanks

OXINARF, 2016-12-20:
That's a good question, I think it shouldn't be sending anything if you are in the correct MAVLINK_TARGETTING mode.
Yes, just adding that line of code to test.

SalahuddinAsh, 2016-12-21:
@OXINARF 
- Regarding the messages Pixhawk was sending, it was my fault. I was sending a wrong message. When I sent the right **DO_MOUNT_CONTROL** to Pixhawk, Pixhawk didn't send any thing to PC.

- I've tested the commit and it worked, kind of. Pixhawk started sending some real values to the PC. But the values of angles are not correct. I mean, I send **4, 8, 16** as roll, pitch and yaw angles from ODroid to Pixhawk, but Pixhawk sent **1, 3, 7** to the PC. 

- I tried to make sure ODroid is sending the right MAVLink messages to Pixhawk. I tried to debug **command_long** message with **DO_MOUNT_CONTROL** and **CMD_NAV_LAND**. The land command is successfully received by Pixhawk and pixhawk go to LAND mode, therefore I took it as a reference. 
Here are some message ODroid sent to Pixhawk. The first set is for **DO_MOUNT_CONTROL** and the other set is for **CMD_NAV_LAND**. I found them strange. As the command_id field in the message is always changing which should not happen, right?

[image]

What do you think about it? What could cause the wrong values of angles? And why the **command_id** field in the **command_long** message is always changing?
Thanks

SalahuddinAsh, 2016-12-21:
@OXINARF
- Regarding the messages Pixhawk was sending, it was my fault. I was sending a wrong message. When I sent the right DO_MOUNT_CONTROL to Pixhawk, Pixhawk didn't send any thing to PC.
- Then I've tested the commit and it worked, kind of. Pixhawk started sending some real values to the PC but the values of angles are not correct. You can see in the picture what ODroid sends and what RPi receives.
- I tried to make sure ODroid is sending the right MAVLink messages to Pixhawk. I tried to debug **command_long** message with **DO_MOUNT_CONTROL**. I connected ODroid directly to the PC and here is what ODroid sends
[image]
I have no idea why param1, 2 and 3 have these values. Do you have any idea? I think this is the main cause of the incorrect angles' values received by RPi, right?

SalahuddinAsh, 2016-12-24:
I finally figured out the missing part. I should cast hex to float on the RPi side to get the right values. I did this using this snippet of code
```
  char buffer[] = {0x00, 0x00, 0x80, 0x41};
  float f = *(float *)&buffer;
  printf("%f\n",f);
```

closed 2016-12-24 by SalahuddinAsh

OXINARF, 2016-12-26:
@SalahuddinAsh Sorry for not answering before, Christmas week is always full. I'm glad you got it working. So you confirm that the commit in #4420 is needed to make MAVLink mode working?