ArduPilot/ardupilot #5421: ROVER : Missing GPS and EKF failsafe

khancyr, 2016-12-20:
#### Issue details
Reproducible with SITL : set a mission, launch a mission, use PARAM set SIM_GPS_DISABLE 1 or PARAM SET SIM_GPS_NUMSATS 0 . The rover will reach the last waypoint because the bearing is good and after get lost and go weird

#### Version
from https://github.com/ArduPilot/ardupilot/commit/802206baede8ec6fe0b6f7896e3efb1b5875d16c and surly long before that

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover

OXINARF, 2016-12-20:
@gmorph I've marked this has a bug as I couldn't find an EKF failsafe and I think that indeed Rover should check the EKF health. Feel free to change the label though.

gmorph, 2016-12-21:
Ok thanks - I'll have a look.

rmackay9, 2019-03-13:
this has been included in Rover-3.5 so closing

closed 2019-03-13 by rmackay9