ArduPilot/ardupilot #5424: Incorrect Altitude being reported in ADS-B Mavlink Msg's

jwuavionix, 2016-12-20:
In mavlink 2's ADS-B send_dynamic_out msg, the GNSS Altitude is being reported in decimeters instead of the correct mm's.  

Also, the pressure Altitude isn't implemented and is always being sent as invalid.

Arduplane V3.7.1 and Arducopter 3.4.3 on a pixhawk

magicrub, 2016-12-20:
Hi Jeff, 

Here's the problem code: 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_ADSB/AP_ADSB.cpp#L472

And 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_ADSB/AP_ADSB.cpp#L519

The GPS alt unit is an easy fix, my bad! The pressure alt one is less obvious. I did not implement it because it wasn't clear what the data was. 

I can fix the GPS alt right away, would you be able to supply the description of the pressure sensor data?

magicrub, 2016-12-20:
GPS alt is fixed via https://github.com/ArduPilot/ardupilot/commit/372837b959db8a2a59a2a92dfbef73d3afd6aedd

jwuavionix, 2016-12-20:
The pressureAltitude reported should be calculated using a standard reference pressure of 1013.25mbar.  I've attached a little doc that describes the calculation which is already referenced to this reference.
[pressureAltitude.pdf](https://github.com/ArduPilot/ardupilot/files/664815/pressureAltitude.pdf)

magicrub, 2017-02-20:
@jwuavionix I just noticed this is still lingering. Please assist on what the data is supposed to be, I want to make sure it's correct.

per the [mavlink packet](https://github.com/ArduPilot/mavlink/blob/e08439ef3d4a98dd9482d446ede78eaf1633f109/message_definitions/v1.0/uAvionix.xml) note: <field type="int32_t" name="baroAltMSL">Barometric pressure altitude relative to a standard atmosphere of 1013.2 mBar and NOT bar corrected altitude (m * 1E-3). (up +ve). If unknown set to INT32_MAX</field>
      
So is it feet above sea level?

jwuavionix, 2017-02-21:
I left a comment on Dec 20th that attached an equation that is used to calculate the standard pressure altitude from the compensated(temp) baro pressure.  This isn't AGL or MSL and should be calculated using the equation attached and just the baro pressure.

magicrub, 2017-02-21:
is it OK if it is temperature compensated?

jwuavionix, 2017-02-21:
yes, the equation needs to use temperature compensated baro pressure as the input.

magicrub, 2017-02-21:
great! Then I think this ends up being easier than we thought. I was thinking into it too much. Thanks @jwuavionix !

magicrub, 2017-02-21:
See PR https://github.com/ArduPilot/ardupilot/pull/5761

closed 2017-02-22 by magicrub