ArduPilot/ardupilot #5427: Some Programming  Bug in calcGpsGoodForFlight(void) for calculate ekfInnovationsPass

longshike, 2016-12-21:
#### Issue details
Here is the part of code in calcGpsGoodForFlight(void) 

// Apply a threshold test with hysteresis to the normalised position and velocity innovations
    // Require a fail for one second and a pass for 10 seconds to transition
    if (lastInnovFailTime_ms == 0) {
        lastInnovFailTime_ms = imuSampleTime_ms;
        lastInnovPassTime_ms = imuSampleTime_ms;
    }
    if (velTestRatio < 1.0f && posTestRatio < 1.0f) {
        lastInnovPassTime_ms = imuSampleTime_ms;
    } else if (velTestRatio > 0.7f || posTestRatio > 0.7f) {
        lastInnovFailTime_ms = imuSampleTime_ms;
    }
    if ((imuSampleTime_ms - lastInnovPassTime_ms) > 1000) {
        ekfInnovationsPass = false;
    } else if ((imuSampleTime_ms - lastInnovFailTime_ms) > 10000) {
        ekfInnovationsPass = true;
    }

when velTestRatio or posTestRatio over 1.0f  for a specific  time, the velTimeout or posTimeout will be true in FuseVelPosNED(), and when veTimeout or posTimeout be true, the velTestRatio or posTestRatio will be set zero, this part code is here

if (velHealth || velTimeout) {
                velHealth = true;
                // restart the timeout count
                lastVelPassTime_ms = imuSampleTime_ms;
                // If we are doing full aiding and velocity fusion times out, reset to the GPS velocity
                if (PV_AidingMode == AID_ABSOLUTE && velTimeout) {
                    // reset the velocity to the GPS velocity
                    ResetVelocity();
                    // don't fuse GPS velocity data on this time step
                    fuseVelData = false;
                    // Reset the normalised innovation to avoid failing the bad fusion tests
                    velTestRatio = 0.0f;
                }
            } else {
                velHealth = false;
            }

so why velTestRatio or posTestRatio compare to 0.7f in here
  if (velTestRatio < 1.0f && posTestRatio < 1.0f) {
        lastInnovPassTime_ms = imuSampleTime_ms;
    } else if (velTestRatio > 0.7f || posTestRatio > 0.7f) {
        lastInnovFailTime_ms = imuSampleTime_ms;
    }
I think it is no meaning here, we can just write like this here 
if (velTestRatio < 1.0f && posTestRatio < 1.0f) {
        lastInnovPassTime_ms = imuSampleTime_ms;
    } else  {
        lastInnovFailTime_ms = imuSampleTime_ms;
    }

or there is some thought that not described here about how compare to 0.7f

#### Version
the master Version

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Copter X

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

eastmoutain, 2016-12-21:
Well, I think the primary idea for the code is to set strict check for both `velTestRatio` and `posTestRatio`. 
1. if the history status is healthy, and both `velTestRatio` and `posTestRatio` are reasonable, then update  `lastInnovPassTime_ms`, else we should consider it is unhealthy, and update `lastInnovFailTime_ms` .
2. if the history status is unhealthy, and either  `velTestRatio`  or `posTestRatio` are larger than 0.7, we shouldn't accept them untill they are both below 0.7.

The logic maybe looks like the code below:
```
static bool ratio_is_healthy = false;
if (ratio_is_healthy) {
    if (velTestRatio < 1.0f && posTestRatio < 1.0f)
        lastInnovPassTime_ms = imuSampleTime_ms;
    else {
        ratio_is_healthy = false;
        lastInnovFailTime_ms = imuSampleTime_ms;
    }
} else {
    if (velTestRatio > 0.7f || posTestRatio > 0.7f)
        lastInnovFailTime_ms = imuSampleTime_ms;  
    else {
        ratio_is_healthy = true;
        lastInnovPassTime_ms = imuSampleTime_ms;
    }
}
```

OXINARF, 2016-12-21:
@longshike This is really a development support question, not a confirmed bug or a feature request, which is the purpose of this issues list. Next time, please post in the forum (http://discuss.ardupilot.org) or ask in Gitter (http://gitter.im/ArduPilot/ardupilot).

Regarding your question, the code is simple: if both ratios are less than 1.0 then everything is OK, but if just one goes above 0.7 then there is something wrong. This is done together with checking how long it has passed since lastInnovPassTime_ms/lastInnovFailTime_ms was changed. The general flow is:

- it just takes one full second of, at least, one of the ratios being over 1.0 to mark the EKF innovations as failed
- it takes 10 full seconds of both ratios to be below 0.7 to go back to EKF innovations marked as pass

This basically means, one ratio above 1.0 should rapidly mark a failure but to return to success both ratios have to be below 0.7 for a longer period of time. Why? Because false success is much worse than false failure - so it is best to be certain of success and quickly turn to failure at "minimum" sight of problems.

closed 2016-12-21 by OXINARF