ArduPilot/ardupilot #5428: EKF: GPS failure doesn't trigger EKF failsafe (master)

rmackay9, 2016-12-21:
There appears to be a bug in master (not in AC3.4) that leads to the EKF failsafe not triggering if the GPS is lost.

The issue can be recreated by doing this in SITL:

- wp load ../Tools/autotest/copter_mission.txt
- LOITER
- arm throttle
- rc 3 1100
- AUTO
- param set SIM_GPS_DISABLE 1

What should happen is after about 10 seconds the EKF failsafe will trigger and the vehicle will switch to LAND mode.  In Copter-3.4 this is what happens.  In master however the vehicle flies off.

The issue seems to have appeared with the addition of the Beacon/Pozyx code on Nov 30th somewhere between these two commits:
- this commit from Nov 30th doesn't work: 74c9c4aa9dabf6985f875f79024be845c5f958ff
- this earlier commit from Nov 30th works: 8af192ed9d1623f44eb492c5b4be392861b3a507

OXINARF, 2016-12-21:
Here is the comparison between the two: https://github.com/ArduPilot/ardupilot/compare/8af192ed9d1623f44eb492c5b4be392861b3a507...74c9c4aa9dabf6985f875f79024be845c5f958ff

There are quite a few changes on the EKF2, at first sight I couldn't find any bug. Almost certain that the bug isn't in the beacon stuff, but in the changes to position timeout/aiding mode checks.

EShamaev, 2016-12-21:
I think the problem is in file AP_NavEKF2_Control.cpp where you rely on beacon to be present to decide if PV_AidingMode is OK.

https://github.com/ArduPilot/ardupilot/compare/8af192ed9d1623f44eb492c5b4be392861b3a507...74c9c4aa9dabf6985f875f79024be845c5f958ff#diff-fb93b6a210716d77fac57f0c7c54f79cR168

So the GPS health gets replaced by beacon and AID_ABSOLUTE stays armed when using beacon even when GPS is lost.

https://github.com/ArduPilot/ardupilot/compare/8af192ed9d1623f44eb492c5b4be392861b3a507...74c9c4aa9dabf6985f875f79024be845c5f958ff#diff-fb93b6a210716d77fac57f0c7c54f79cR212

tridge, 2016-12-21:
this is fixed in master, thanks to Paul

closed 2016-12-21 by tridge