ArduPilot/ardupilot #5430: Rover: make use of AP_Motor

khancyr, 2016-12-21:
#### Issue details
Rover is currently well working in manual mode in traditionnal rover ( 1 output for power, 1 output for steering). Skid steering mode could be improvement, and auto mode don't have the full potential of Copter one for example.

In order to bring more improvement and unified the code base, it could be interesting to make an exension of AP_Motor for rover, as traditional helicopter already use it, the work shouldn't be too hard.
This will unlock the usage of one-shot pwm mode and PWM motor (A special attention must be made to link another pin or chip on spi/i2c bus to the motor output to control the rotation direction. In fact, most H-bridge use a pin for direction selection)
This will also permit to complex rover with more wheel, or exotic equipment (jumper, robotics arm etc.)

R-Lefebvre, 2016-12-30:
I think the idea of starting to combine these classes is a good thing in our march toward unification of the code.

However, please don't base this work on the name "Motors" which is totally wrong in this context.  It should never have been called AP_Motors, because it makes no sense for helicopters where servos are being driven, etc.

Sorry for being pedantic but this has bothered me for years.  I have to imagine it leads to confusion for new contributors surfing the code.

The class should be called AP_Actuators or something similarly hardware agnostic.

rmackay9, 2017-01-04:
I certainly agree Rover should at least get a new class that separates the output mixing from the rest of the vehicle code.  It should make the architecture much easier to understand.

khancyr, 2017-07-18:
Done in https://github.com/ArduPilot/ardupilot/pull/6547

closed 2017-07-18 by khancyr