ArduPilot/ardupilot #5431: Rover: make use of Internial nav

khancyr, 2016-12-21:
#### Issue details
Currently, rover use AHRS , a switch to inertial nav could permit (if I understand correctly the difference) more precise behaviour in case of weak gps.
This will also unlock the works done on copter for avoidance

OXINARF, 2016-12-21:
We have talked before about merging AP_InertialNav into AP_AHRS because the first is a very tiny layer now that has little purpose as a separate library. Given that, I think it would make more sense for someone to do that work and then Rover could take advantage of any new APIs in AP_AHRS.

amilcarlucas, 2017-09-25:
There is a PR now to remove inertialNav from Copter. Should this be closed ?

peterbarker, 2017-12-19:
@amilcarlucas I think so.  Let's see if anyone notices.

closed 2017-12-19 by peterbarker

gmorph, 2017-12-19:
I noticed - but I agree that it should be closed :-)