ArduPilot/ardupilot #5434: Running SITL inside Vagrant doesn't start MAVProxy properly

danielschonfeld, 2016-12-21:
You base the `--out` argument input based on the username, which in `ubuntu/yakketi` is `ubuntu` and not `vagrant`

See the following line in the code:
https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L659

peterbarker, 2016-12-24:
@danielschonfeld I'd love to see a PR, thanks!

peterbarker, 2016-12-24:
@danielschonfeld Might be good to find a different proxy for "running under Vagrant" than checking the username; we're at the whim of the provider of the box as to what username we end up with.

danielschonfeld, 2016-12-24:
@peterbarker why don't we change the `Vagrantfile` to include a private network in which the host IP will be static then we can add it as an additional `--out` flag to MAVLink?

That way, if its under our vagrant environment it works, and if its not its just another `--out` flag that doesn't harm anything if it's under a different environment.

peterbarker, 2016-12-24:
On Sat, 24 Dec 2016, Daniel Schonfeld wrote:

> @peterbarker why don't we change the Vagrantfile to include a private
> network in which the host IP will be static then we can add it as an
> additional --out flag to MAVLink?
>
> That way, if its under [...]

That would typically lead to a "can't assign requested address" error, so 
we probably still need a "running_under_vagrant()" call (checking for 
existence of /vagrant would be a quick fix for that)

Statically defining the network gives (more) opportunity to conflict with 
other network definitions.  OTOH, the current system where we simply 
assume we're on 10.0.2.x/y is pretty bodgy!  I guess ideally we would ask 
MAVProxy to bind to an interface rather than an address.  However - that's 
a different problem for a different day!

danielschonfeld, 2016-12-24:
@peterbarker what about touching a file in the Vagrant provisioning shell file, something like `touch /tmp/vagrant.env` and check for it inside `sim_vehicle.py`? that would be more reliable way ?

I can submit that PR if you like this idea easily

peterbarker, 2016-12-25:
On Sat, 24 Dec 2016, Daniel Schonfeld wrote:

> @peterbarker what about touching a file in the Vagrant provisioning shell
> file, something like touch /tmp/vagrant.env and check for it inside
> sim_vehicle.py? that would be more reliable way ?

That would also work, but perhaps ${VAGRANT_USER_HOME_DIR}/.is_vagrant 
would be more appropriate, as /tmp may get nuked at any time for any 
reason.

Does this have any particular advantage over checking for /vagrant, 
'though?  It is unlikely someone will have that directory present on the 
host machine, and if tht were to change then there will be more than a few 
miffed people out there.

peterbarker, 2016-12-29:
Merged relevant PR; eventually a file in / was used.

closed 2016-12-29 by peterbarker

danielschonfeld, 2016-12-29:
@peterbarker sorry i got busy with the holiday.  if you prefer it to be checking for `/vagrant` i can happily submit another PR

peterbarker, 2016-12-30:
@danielschonfeld Too late now!  Going to have to wait until this way maims someone :-)

Thanks!