ArduPilot/ardupilot #5437: GPS_Type inconsistent with MAVLINK GPS_FIX_TYPE regarding RTK and does not differentiate between RTK FLOAT and RTK FIX

rrr6399, 2016-12-22:
#### Issue details
I've been working with the Emlid Reach recently and notice that when it gets a RTK FLOAT FIX, its being reported as a DGPS fix instead.  It appears that the AP_GPS.h code does not differentiate between RTK FLOAT and RTK FIXED:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.h#L76
   enum GPS_Status {
        NO_GPS = 0,             ///< No GPS connected/detected
        NO_FIX = 1,             ///< Receiving valid GPS messages but no lock
        GPS_OK_FIX_2D = 2,      ///< Receiving valid messages and 2D lock
        GPS_OK_FIX_3D = 3,      ///< Receiving valid messages and 3D lock
        GPS_OK_FIX_3D_DGPS = 4, ///< Receiving valid messages and 3D lock with differential improvements
        GPS_OK_FIX_3D_RTK = 5,  ///< Receiving valid messages and 3D lock, with relative-positioning improvements
    };
Here's the MAVLINK definitions:
          <enum name="GPS_FIX_TYPE">
              <description>Type of GPS fix</description>
              <entry value="0" name="GPS_FIX_TYPE_NO_GPS">
                  <description>No GPS connected</description>
              </entry>
              <entry value="1" name="GPS_FIX_TYPE_NO_FIX">
                  <description>No position information, GPS is connected</description>
              </entry>
              <entry value="2" name="GPS_FIX_TYPE_2D_FIX">
                  <description>2D position</description>
              </entry>
              <entry value="3" name="GPS_FIX_TYPE_3D_FIX">
                  <description>3D position</description>
              </entry>
              <entry value="4" name="GPS_FIX_TYPE_DGPS">
                  <description>DGPS/SBAS aided 3D position</description>
              </entry>
              <entry value="5" name="GPS_FIX_TYPE_RTK_FLOAT">
                  <description>RTK float, 3D position</description>
              </entry>
              <entry value="6" name="GPS_FIX_TYPE_RTK_FIXED">
                  <description>RTK Fixed, 3D position</description>
              </entry>
              <entry value="7" name="GPS_FIX_TYPE_STATIC">
                  <description>Static fixed, typically used for base stations</description>
              </entry>

The reason it matters is that the logic used to pick the best GPS is based on the GPS status value. The RTK Float solution will not be used compared to the DGPS solution of the platform's first Ublox GPS even though it is more accurate even if it has less satellites. 
https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp#L516

Please add the RTK FIXED status and differentiate between DGPS, RTK FLOAT and RTK FIX so that the accurate fix type will be reported to the user and that the GPS selection logic will be improved.

#### Version
3.4 and later

#### Platform
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk and Emlid Reachh

WickedShell, 2016-12-22:
> libraries/AP_GPS/AP_GPS.h#L76
> libraries/AP_GPS/AP_GPS.cpp#L516

This is partially addressed in the m8p branch. I refractored all GpS types
to report RTK if they support it, but if the code wasnt clear if it was a
float or a fixed type I just called it float

WickedShell, 2017-01-25:
Fixed in #5498 and #5616 is an addon patch for the GSOF driver.

closed 2017-01-25 by WickedShell