ArduPilot/ardupilot #5448: Feature: Replace sd card by emmc

jinchengde, 2016-12-27:
#### Issue details
we had try to connect the CMD, CK, D0~D4 to emmc BGA pin, and also include VSS, VDDF, VDDI, but failed to startup.

Then we try a small pcb board transfer emmc to sd, we connect the sd card to computer, copy file to sd, everything is ok, but when we connect CMD, CK, D0~D4 from pixhawk to sd, it failed again.

At last, we connect the pixhawk to tf card directly, it could startup, no problem.

So I suppose it maybe some wrong from emmc to SD, maybe something different in STM32 or Nuttx in the driver, Does anyone have experience to do that?

#### Version
3.6

#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
fixed wing

#### Hardware type
pixhawk

#### Logs
null

magicrub, 2016-12-27:
Have you tried the forums yet? HTTP://discuss.ardupilot.org

jinchengde, 2016-12-27:
http://discuss.ardupilot.org/t/replace-sd-card-by-emmc/13634/4
but，maybe I should be in PX4 project?

jinchengde, 2017-01-01:
That was solved by change another emmc chip.

magicrub, 2017-01-01:
Great!

closed 2017-01-01 by magicrub