ArduPilot/ardupilot #5450: Feature request: enhance Circle mode

night-ghost, 2016-12-27:
Circle mode as it is now is rather useless - the pilot is not able to control the  diameter of the circle and the speed of copter besides via GroundStation and parameters. So for now flight by the converging or diverging spiral is impossible

It is proposed to allow the pilot to control the speed, diameter and the center position.

To do this, new function for channel 7/8 requred - "circle control". Whereas this mode deactivated right stick should move the circle's center relative to pilot, as it does in Loiter with SuperSimple for copter itself. Whereas mode is activated, right stick forward/back should adjust circle diameter where left/right adjust copter speed.

The resulting control seems to be fairly intuitive both in visual and FPV, and significantly improve the usability of  this mode.
PS. Inspired by Yuneec Q500 Typhoon circle mode

#### Version
git

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-12-28:
It's a great idea.  We've had similar discussions of this before actually:

- allow repositioning: https://github.com/ArduPilot/ardupilot/issues/386
- allow changing radius with ch6: https://github.com/ArduPilot/ardupilot/issues/876

I think we could get away without having a new auxiliary switch feature.  Instead we could just consume the roll and pitch inputs whenever we're in the circle flight mode.  We wouldn't consume the pilot inputs during a mission's LOITER_TURNS command.

night-ghost, 2016-12-28:
>I think we could get away without having a new auxiliary switch feature 

Without switch it's hard to select between radius+speed and reposition modes. To change mode to loiter for reposition is not the same because it breaks circle movement

geofrancis, 2017-02-06:
@rmackay9 that sounds like a better way of doing it, pitch could control the size of the circle and and roll could control the speed and reverse the circle.

rmackay9, 2017-02-06:
@geofrancis, txs for the feedback.  yes, I think using roll-and-pitch could be good.  currently those sticks aren't used in circle so it's "efficient" in some way to use them instead of having to setup other switches/knobs.

hyperion11, 2017-02-06:
Pitch stick i think it will be most convenient to change radius.

shortstheory, 2018-03-14:
I am interested in working on this issue.

rmackay9, 2018-03-14:
@shortstheory, txs for considering taking this on.  One of the main things we want to do in 2018 is increase our team's developer capacity.

So I think the idea is the pitch stick should increase/decrease the circle's diameter and maybe roll increases the vehicle's speed around the circle.

I'd start with the more obvious pitch-stick interpretation.  On each iteration you could check the position of the pitch stick and then call the copter.circle_nav.set_radius().  There are a few things to be careful of:

- need to decide what the maximum rate that the circle radius can increase/decrease.  it might be good to use the wpnav's loiter speed as the maximum... or maybe this is too high, best to test in the simulator and see.
- check how the vehicle behaves (in the simulator) as the circle radius is increased/decreased.  it will probably be smooth because the navigation controller is being used but be on the look out for the copter being unable to keep up with changes to the circle radius.
- the main loop rate can be modified by the user so you can't assume it's 400hz.  there's a scheduler method that return the main loop rate.

geofrancis, 2018-03-16:
Ideally roll should control the speed of the circle.

rmackay9, 2020-02-03:
I think @Pedals2Paddles has implemented this recently and it's included in Copter-4.0.1 so closing, thanks!

closed 2020-02-03 by rmackay9