ArduPilot/ardupilot #5451: Copter: EKF lane changing if started with no compass

rmackay9, 2016-12-28:
We've had one reported case where altitude resets were occurring with AC3.4.2 in AltHold mode when the board was configured to not use any compass (i.e. COMPASS_USE=0).

@priseborough says The EKF was not designed to start without a magnetometer for copter applications, so I need to tighten up the status reporting so that it reports the attitude as unhealthy if the yaw angle has not been initialised from the magnetometer or some other source if the vehicle is not a fly-forward type of vehicle (eg a plane) that can do an in-flight yaw alignment using the GPS velocity

priseborough, 2016-12-28:
I have modified EKF3 to fly without magnetometer for copters and flown it without optical flow in alt_hold and with optical flow in loiter mode.

https://github.com/ArduPilot/ardupilot/pull/5407/commits/52e9e429f08c80c989414e526a8b07e90860832a

This has been included in PR#5407

wyq200704, 2016-12-29:
In AC3.2.1 i can arm the ardupilot,but in AC3.4.3 disable the compass, ardupilot cannot be armed.Mp reported compass not healthy. Maybe This pr fix it.

rmackay9, 2016-12-29:
@wyq200704, thanks for the report.  it's unlikely that the change from Paul will resolve that.  Most people should be flying with a compass but it makes sense, as you suggest, that we don't trigger the compass pre-arm check if the user has specifically disabled all the compasses.  I might raise a separate issue for this.

geofrancis, 2017-02-06:
now the cheap RTK gps systems are getting more popular what about supporting 2 as compass, now that with ekf3 that you can set where on the machine they are.

priseborough, 2017-02-06:
I would add the calculations for yaw to the AP_GPS library and add a public interface providing the yaw and estimated accuracy. We already have the ability to set the body frame position for each GPS, so all the data required is there.

proficnc, 2017-02-06:
This makes me very happy :) 

Really looking forward to this!

proficnc, 2017-02-06:
Should this be its own issue Paul?

geofrancis, 2017-02-06:
https://github.com/ArduPilot/ardupilot/issues/3934

rmackay9, 2017-02-07:
I think we can actually close this issue.  The latest comments above are really about a new issue which is the ability to use two GPSs in place of a compass.

I've created a new issue to cover this request here: https://github.com/ArduPilot/ardupilot/issues/5677

closed 2017-02-07 by rmackay9

R-Lefebvre, 2017-02-07:
I don't think this should have been closed, as these are two separate issues.

Using two GPS's is just one way of flying without a compass.  It's not the only solution, and Paul was tinkering with another idea.

rmackay9, 2017-02-07:
@R-Lefebvre, I think though, it's "issue creep".  So the original issue that I raised was addressed by Paul in PR#5407.  I see the two GPS issue which we've got a new issue for now.  I think if there are other ideas or enhancements, it's fine to put in a new issue.