ArduPilot/ardupilot #5454: Copter: don't trigger pre-arm compass check if user has disabled all compasses

rmackay9, 2016-12-29:
Normally users will fly with a compass, but if they've specifically disabled all compasses by setting COMPASS?_USE to zero then we should not trigger the compass pre-arm checks.  Of course, we might also not enable position hold modes like Loiter, etc.

This is related to this other issue: https://github.com/ArduPilot/ardupilot/issues/5451

EbinPhilip, 2018-03-08:
Hi! I'm new to the ardupilot and codebase would like to work on this issue. I couldn't find this COMPASS_USE parameter in parameters.h or ap_compass, but i did notice a parameter compass_enabled. Is that what you were talking about?

khancyr, 2018-03-08:
Hello,
The parameter is in ap_compass : https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass.cpp#L91

EbinPhilip, 2018-03-08:
I went through compass_checks() in AP_Arming. It already checks if COMPASS_USE is set to 0 and skips.check:
https://github.com/ArduPilot/ardupilot/blob/1baa6eb049ab7090b845ffe0b50c7978c1fb9e73/libraries/AP_Arming/AP_Arming.cpp#L279
So I have added some code in mode.cpp to disable GPS based flight modes(since they need compass too) if COMPASS_USE is 0. I have tested it in SITL and it works fine. Can you please take a look?
https://github.com/EbinPhilip/ardupilot/blob/88e3119d0bb6ccb96b0dbce191a8e634c32bace3/ArduCopter/mode.cpp#L195
Shall i open a pull request? 
About the MAG_ENABLE param, #5549 says its a redundant thing. Should I add that right now or shall i leave it for now, since its a seperate issue?

dwiel, 2019-07-23:
I think I am having this issue. One thing to note however is that I am providing pose via SLAM so guided and loiter modes should still be allowed.

rmackay9, 2019-07-23:
@dwiel, yes, I think this issue still exists..

peterbarker, 2020-06-18:
This has now been addressed as part of the external-yaw patches.

closed 2020-06-18 by peterbarker