ArduPilot/ardupilot #5456: Feature request: return by own tracks in RTH

night-ghost, 2016-12-29:
Sometimes flights are held in difficult conditions - mountain, forest, city, so returning back in a straight line is road to disaster.

It is proposed a RTH mode when the forward route  memorized  (using the algorithm dropping extra points and selecting key points, this algorithm is known and tested),  and do returns on this route after dropping the rings and reversing. In this case, the return will be held in guaranteed safe passage.

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover

OXINARF, 2016-12-29:
This is the same principle as #1571, so closing this one.

closed 2016-12-29 by OXINARF

night-ghost, 2016-12-29:
What speed of closing! 

No, this DEFINITELY is not the same, moreover that issue is still not realized. I not only offer an idea but have a way to realize it because I have tested all required algorithms

OXINARF, 2016-12-29:
How the idea different?

> moreover that issue is still not realized

If it was done it would be closed...

> I not only offer an idea but have a way to realize it because I have tested all required algorithms

Great, where's the PR?

night-ghost, 2016-12-29:
>How the idea different?

not to record FULL track but only keypoints - this allows to pack a long journey to small memory

>Great, where's the PR?

the same place where the previous ones. I  will to waste my time on them when at least one will be adopted.

peterbarker, 2016-12-29:
On Thu, 29 Dec 2016, Francisco Ferreira wrote:

> This is the same principle as #1571, so closing this one.

'though a reference for optimizing the return path for space would be 
nice.

rmackay9, 2016-12-29:
I've added one algorithm that was discussed previously to the other PR.  I agree we don't need two PRs open for this one item and normally we keep the older one.

OXINARF, 2016-12-30:
> not to record FULL track but only keypoints - this allows to pack a long journey to small memory

That's an implementation detail that doesn't deserve a new issue. You're very welcome to share your ideas on how to implement a safe RTL in the other issue.

> the same place where the previous ones. I will to waste my time on them when at least one will be adopted.

I'm confused. Is it in the same place as the previous ones or not? Because I don't see any new PR from you. As for the two before, they weren't merged as you haven't followed most of our rules of contribution - even then they are still open as the contributed features were considered interesting.

@peterbarker I'm sure there are other issues opened where that has been discussed, I referenced the first appropriate open issue I could find. In any case, it can always be added to any related open issue, no need for a new one.

night-ghost, 2016-12-30:
my role to offer - your to refuse :) The same ideas are offered as well in INAV project, so can see where they will be implemented first - by a team of professionals or by one enthusiast.

My track reducing algorithm can be seen in action on site [ykoctpa.ru](http://www.ykoctpa.ru/map/?prov=google&z=8&ll=54.9476355830555%2C52.632673443697726&maptype=roadmap&mh=640&on=track) , just change map's scale

peterbarker, 2016-12-30:
On Thu, 29 Dec 2016, night-ghost wrote:

> my role to offer - your to refuse :) The same ideas are offered as well in
> INAV project, so can see where they will be implemented first - by a team of
> professionals or by one enthusiast.
> 
> My track reducing algorithm can be seen in action on site ykoctpa.ru , just
> change map's scale

Sorry, I'm very confused here.

You created an issue; someone pointed out that it seemed to be a duplicate 
issue (it initially seemed to be a request for a feature to be implmented, 
and closed this duplicate issue.

However, it has become clear that you have working code!

Is there some reason you created an issue to track things rather than a 
pull request?

I'm sure we'd all love to see a pull request!

night-ghost, 2016-12-31:
>Is there some reason you created an issue to track things rather than a pull request?

Because I can make a library which collect track with point reduction  only - but not full controller which uses it for RTH needs

rmackay9, 2016-12-31:
@night-ghost, I'm worried that such a library would eat an more memory than the Pixhawk can support, but I'm sure the linux boards could handle it.  If you implement such a library, I'm happy to try and integrate it into RTL.

night-ghost, 2016-12-31:
>I'm worried that such a library would eat an more memory than the Pixhawk can support

Don't worry, code size is rather small, and memory required to store track itself can be just fixed array: when the number of points reaches the size of the array, the algorithm can increase the margin of error in order to weed out the extra point. 
After moving stack to CCM in Revomini it have  only near 37K of used RAM so more than 80K free. If track memory uses  a half of it - 40K - it can hold 5000 points of track. The longest track on my site has 3712 points and it is a full day journey in the impassable forest in the mountains.

night-ghost, 2016-12-31:
PS. All my wishes are aims to one goal.

Somewhere in the mountains, in 50-200km from infrastructure,  there are different points of interest, such as waterfalls. The best time of observation - the period of snowmelt, but to get to them we need the whole expedition from the Army's SUV and then walking from 6 to 25 kilometers through the jungle. And I already 4th time I skip the right time,  coming sooner or later.

Therefore, there is a desire to monitor the state by a drone. But it is necessary or to fly directly through the mountains with the rise and descent of 1 km that is unprofitable for the battery or on the route along the river. And RTH in this case is a guaranteed disaster.

That's why I want an airplane's flight with copter's hovering, cost-effective descent, work in the lightest FC, and other other.

lvale, 2016-12-31:
I've had a test some time ago on this problem, and one of the first issues was the limited storage of "safe" points, so I've created with the help of Tridge a PR to increase the available memory reserved for that https://github.com/ArduPilot/ardupilot/pull/4082 which was then closed and this https://github.com/ArduPilot/ardupilot/pull/4172 was created
Other option was to store waypoints on the sd card and dynamically loading points to memory. 
The algorithm to determine which points to save was rather crude and consuming of cpu resources because it would compare spheres with a calculated radius and the radius was a function of vehicle speed with a minimum value of 5m (user adjustable parameter) and would compare with all the points on the array from latest to older and if a comparison was found would simply reset the index to that and keep recording. When a smart RTH was initiated it would be a simple mission with the points being sent in reverse order.

night-ghost, 2016-12-31:
>The algorithm to determine which points to save was rather crude and consuming of cpu resources 

You say that this is theoretical impossibility - or about problems of implementation specific to your algorithm? Just avoid quadratic complexity!

My measurements shows that the main thread spent 70% of his time in a variety of delay() so if to use this time that resources quite enough.

lvale, 2016-12-31:
@night-ghost My implementation was basic, so that could be an issue (I'm sure it was) and also I was doing a 3D calculation of distance between the center of 2 spheres with a variable radius (the radius was a function of velocity vector (3D)). Then I would go on each waypoint (I was using the existing structure of waypoints) and repeat the calculations. After about 400 points stored the cpu overruns were too much.
I'm sure that a clever c++ implementation would be feasible, but unfortunately real life work took a priority :)

night-ghost, 2016-12-31:
I definitely feel quadratic complexity :) I worked on this algorithm a few years, and now the convolution of the raw track "Vladivostok, Kaliningrad," with the size of several megabytes takes a fraction of a second on desktop. And yes, no 3D calculations because very small chance that wee need to be round the world itineraries

night-ghost, 2017-01-05:
Where can I get coordinates of UAV - not ones from GPS but already filtered by EKF?

lvale, 2017-01-05:
on location.cpp

```
// return distance in meters between two 3D locations
float get_distance3D(const struct Location &loc1, const struct Location &loc2)
{
    float dlat = (float)(loc2.lat - loc1.lat);
    float dlong = ((float)(loc2.lng - loc1.lng)) * longitude_scale(loc2);
    float dheight= ((float)(loc2.alt-loc1.alt))*0.01f;    //Alt comes in cm
    return pythagorous3(dlat, dlong, dheight) * LOCATION_SCALING_FACTOR;
```

night-ghost, 2017-01-05:
I'm sorry, I can misunderstand but this function calculates the distance between two points. But I need to get the current coordinates of the UAV itself at a specified interval to store as track

rmackay9, 2017-01-06:
@night-ghost, either the AHRS object or in copter you can use the inertial_nav object.  In copter we mostly use offset-from-ekf-origin while in plane, rover it's lat/lon/alt.  If you're only implementing for Copter, it's recommend using hte offset-from-ekf-origin.  I think either AHRS or inertial nav can provide that but if you get it from the AHRS it will probably be in NED while from intertial nav it will be NEU.  There's a to-do item to remove inertial-nav but just haven't gotten to it yet.

night-ghost, 2017-01-06:
Ok, thanks. This feature is more required for boat/rover but I can debug it only for copter - so in sample of library I will use copter's data.

night-ghost, 2017-01-16:
Library is almost ready. Tomorrow I'll record a screencast of it's work and it will  be ready to production

rmackay9, 2017-01-17:
very exciting!

night-ghost, 2017-01-17:
https://youtu.be/D5UGfn47i8U

rmackay9, 2017-01-17:
interesting.  it works in 3 dimensions ok?

night-ghost, 2017-01-17:
original video - http://www.ykoctpa.ru/groups/eye-in-a-sky/files/1591/

night-ghost, 2017-01-17:
no only 2D - baro drifts, GPS altitude is very unstable. And no needs for 3d for boats and rovers. So plane/copter should use the same altitude as in RTH

night-ghost, 2017-01-17:
PS. video recorded on STM32F103, time in video (right pane) - in uS.

rmackay9, 2017-01-17:
ok but with the missing altitude though I worry especially for Copter that it won't really be what people expect.

night-ghost, 2017-01-17:
You see, it is unreal that segments will intersect in 3d! So it will be no loops removing. Vertical accuracy should differ from horizontal - so twice calculations. 

May be it is "not what people expect" when we talk about flights in buildings but in real world there are usually no roof on top. And I think that even with fixed altitude this mode can be better than return by direct line.

I can collect altitude of points but exclude it from all calculations - but this will be even less expected than now.

magicrub, 2017-01-17:
Altitude is required.

night-ghost, 2017-01-17:
I am **fundamentally** opposed to the inclusion in the height to the track. If all would have lidar and optical flow - then it would perhaps make sense. But now use the height will result in a huge number of accidents because of the atmospheric pressure changes. Period. 

So there will be no PR - I think that  this feature should be available only by commercial license :)

khancyr, 2017-01-17:
I am in favor of a first version without altitude, as it could be a base for future improvement and it will be a good feature for rover !
@night-ghost If you need testing on real rover, I go some !

OXINARF, 2017-01-17:
I think that having a 2D library for Rover could be a good starting point. The video looks good - I'm no sure it always selects the best path but it is, at least, a first good approach.

For Plane and Copter, I agree with Randy and Tom, altitude is needed. Without it, this would be useful when you have your path perfectly planned as then you can figure out the perfect RTL altitude. But if you are going free flight you are going to configure your RTL altitude as you do now: above all obstacles - that would make the safe path a worse option than a direct one.
I understand the concern about baro drift in Plane, but for Copter I don't think it is very relevant.

magicrub, 2017-01-17:
Having this as a Parameter sounds reasonable

lvale, 2017-01-18:
Great demo @night-ghost 
I believe that even without altitude it's a good starting point for a breadcrumb system.
Please @night-ghost go ahead with the PR :)