ArduPilot/ardupilot #5460: Publish log item meaning.

AndKe, 2016-12-30:
this is based on:
http://diydrones.com/forum/topics/log-interpretation-1

The project has very good information about parameters.
In a similar way, we need to have documentation about logged data, there are lots of logged events, that require require code reading to understand their function, or how to use them for diagnosis.

DonLakeFlyer, 2016-12-30:
In QGC 3.2 we'll be revamping the charting support. Part of that will likely including bringing both telemetry data and log data in the QGC FactSystem. This means full meta data will be available for all these values just like parameters. At least from a technical perspective. In order to make it all work it will require plumbing/creation of meta data for all mavlink messages as well as all log entries.

OXINARF, 2016-12-31:
I understand the need for this, but the log structures change frequently making it hard to keep documentation. The dataflash log was designed as self-describing, which is great, but the variable names aren't very descriptive - and indeed either you already know what a variable is or you have to go code searching.

If there are suggestions on how to achieve this in a maintainable way, they are very welcome.

DonLakeFlyer, 2016-12-31:
Built the log structures from an xml definition as part of the build process. Parse the XML to generate the C++ code. Then also use the xml definitional as the meta data for the logs. Basically the same way mavlink message definitions work.

OXINARF, 2017-03-15:
Unfortunately a new issue for this was started (#5864) and it already has more posts so I'm closing this one.

closed 2017-03-15 by OXINARF