ArduPilot/ardupilot #5461: Switching to AltHold freezes when barometer throws error

daniel-leonard-robinson, 2016-12-30:
#### Issue details
I was descending from height in stabilise mode, until I reached a point where I was traveling at 10m/s and I switched to AltHold to stabilise the hexacopter. Unfortunately the drone froze and plummeted to the ground, reaching about 100 km/h (between 27 and 31 m/s).

I realize that speed does affect barometer readings, however, if an error checking of some sort (besides if switching to a flight mode is successful) could be provided for in flight_modes.cpp in future, that would be great.

In the flight logs, the last logged message was _Barometer_, with error code 2, and that was immediately after switching to AltHold.

#### Version
AC 3.2.1

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover

#### Airframe type
hex

#### Hardware type
APM 2.8

#### Logs
[2016-12-29 14-18-14.zip](https://github.com/ArduPilot/ardupilot/files/679235/2016-12-29.14-18-14.zip)

magicrub, 2016-12-30:
The hardware, and thus the firmware, you are using was depreciated over 2 years ago. Before closing this is, can anyone confirm this problem is already fixed?

OXINARF, 2016-12-31:
The logged error (baro glitch) isn't used anywhere anymore. In any case a logged error shouldn't freeze the software. I have no idea if this was ever a reported and fixed issue or not - I'm not even sure that the freeze is related to the baro error, it may not even be a software issue.

As @magicrub said the used firmware version is very old, much has changed since then. Because this isn't a confirmed bug it should instead be in the forum (http://discuss.ardupilot.org) and so I'm closing this.

closed 2016-12-31 by OXINARF