ArduPilot/ardupilot #5462: Stopped navigating on the first waypoint

JeyPi, 2016-12-31:
The plane did not follow the route. It started to fly to the first waypoint and then it started to hold there. the height during the hold changed nearly 100 meters. I checked the route and in my opinion the plane should had flown to the next waypoint. The gps was OK during the whole flight. I might had encountered a bug

Version
ArduPlane V 3.7.1

Platform: 
Plane

Airframe type
Plane -  elevator, rudder, ailerons, flaps

Hardware type
Pixhawk,

Log:
https://drive.google.com/open?id=0Bw3cHAyxN4dBMnRYOW1OQ1RqRUU

magicrub, 2016-12-31:
I'll take a look at that log tomorrow

JeyPi, 2016-12-31:
It is solved. I have activ the missiom restart function

closed 2016-12-31 by magicrub