ArduPilot/ardupilot #5471: Heli Setup window doesn't works correctly on firmware 3.3.4

Fortesque0, 2017-01-01:
#### Issue details
Hello, 
It's not possible to set up servo reverse and trims using firmware 3.3.4 on traditional helicopter heli setup window. You can check the box and change trim values, but nothing happens.

#### Version
AC 3.3.4

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Traditional helicopter 450

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk Hkpilot32
#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-01-03:
Is this an issue with ArduPilot or an issue with Mission Planner?

Fortesque0, 2017-01-03:
I think this is an issue of ardupilot. I use it with a downgraded MP and same happens. Then, i use last firmware (3.3.3) and works fine on both MP versions

rmackay9, 2017-01-03:
It's more likely that it's a mismatch between mission planner and ardupilot caused by renaming some parameters.  The solution will most likely be to modify mission planner to use the new ardupilot names.

OXINARF, 2017-01-03:
@rmackay9 Yeah, that's what I was thinking.

@Fortesque0 Can you please open an issue in the Mission Planner repo?

Fortesque0, 2017-01-03:
Ok, I will open a new issue in the mission planner.
Thanks!

OXINARF, 2017-01-06:
I'm closing this as it looks like it was indeed an issue with MP that @meee1 has just fixed in master. I'll let the one in MP repo open until @meee1 or @Fortesque0 confirm it is fixed.

closed 2017-01-06 by OXINARF