ArduPilot/ardupilot #5484: Copter 3.5 allow to use gps and optical flow simultaniously

roque-canales, 2017-01-02:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-01-03:
Already done in Copter 3.4. I think the documentation needs to be updated though, can you please create an issue in the wiki repo?

closed 2017-01-03 by OXINARF

rmackay9, 2017-01-04:
Actually I'm not sure this is done in AC3.4.  I think it will fall back from GPS to optical flow if the GPS completely fails but it won't actively switch between them or use both sets of data at the same time.
I think @priseborough can confirm.

OXINARF, 2017-01-04:
It's done for sure: https://github.com/ArduPilot/ardupilot/pull/4933

@rmackay9 you merged it, have you forgot? :smile:

rmackay9, 2017-01-04:
ok then!  so I guess it's just about updating the docs.  I've added it to the list for the AC3.5 release (although really it applies to AC3.4 as well).  https://github.com/ArduPilot/ardupilot/issues/5160

priseborough, 2017-01-04:
I checked out AC 3.4 and the EKF2 should be able to use GPS and flow simultaneously.

priseborough, 2017-01-04:
However this is not something that I have tested on the 3.4 release.

OXINARF, 2017-01-04:
@priseborough Well, you tested your PR, which went into master and from which 3.4 was then build out of. So I think it should all be working.

maciek01, 2018-10-30:
what should the ek2_gps_type be set to so both flow and gps are used? is 0 a good idea? (AC 3.5.7)