ArduPilot/ardupilot #5499: Should add 5.4.1 to the /ardupilot/modules/PX4Firmware/makefiles//toolchain_gnu-arm-eabi.mk:57: *** Unsupported version of arm-none-eabi-gcc, found: 5.4.1 instead of one in: 4.7.4 4.7.5 4.7.6 4.8.4 4.9.3.  Stop.

ivan95603, 2017-01-05:
#### Issue details
_Please describe the problem, or desired feature_

**This is not an issue i guess it is a feature I have arm-none-eabi-gcc 5.4.1 and added it to file so it passed compilation successfully.** 

#### Version
_What version was the issue encountered with_
Current
#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_

/ardupilot/modules/PX4Firmware/makefiles//toolchain_gnu-arm-eabi.mk:57: *** Unsupported version of arm-none-eabi-gcc, found: 5.4.1 instead of one in: 4.7.4 4.7.5 4.7.6 4.8.4 4.9.3.  Stop.
make[2]: Leaving directory '/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build'
/ardupilot/modules/PX4Firmware/Makefile.make:143: recipe for target '/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/firmware.px4' failed
make[1]: *** [/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/firmware.px4] Error 2
make[1]: Leaving directory '/ardupilot/modules/PX4Firmware'
../mk/px4_targets.mk:182: recipe for target 'px4-io-v2' failed
make: *** [px4-io-v2] Error 2

OXINARF, 2017-01-05:
This issues list is for confirmed bugs or feature requests.

I don't know what you are doing, but the officially supported GCC version is 4.9.3 and it works very well.

closed 2017-01-05 by OXINARF

ivan95603, 2017-01-05:
@OXINARF, 
I know but it also works with 5.4.1. But if you are also working with https://github.com/PX4/Firmware you then probably have this version.

OXINARF, 2017-01-05:
@ivan95603 I haven't tried it, but someone in the dev team did and there were some issues, I'm not sure exactly what it was. Anyway, you can have more than one compiler installed.

khancyr, 2017-01-05:
@OXINARF @ivan95603 weird it failed with make but build with waf. 
Try with waf if you really want new gcc but there is no garantie that is will work (on  arm-none-eabi-gcc 6.2.1  it failed). That is because Px4Firmware is not up-to-date with PX4 master.

OXINARF, 2017-01-05:
It's no weird. make uses the old Makefiles system in PX4Firmware, waf uses the new CMake build system.

khancyr, 2017-01-05:
haaaaaaa ok ! then it make sense! ;-P