ArduPilot/ardupilot #5503: Internal (SPI) magnetometers occasionally fail to sample

jschall, 2017-01-05:
#### Issue details
This is a plot over 15 hours of flight:
[image]
The internal (SPI) magnetometers occasionally fail to sample. The external (i2c) mag does not.

#### Version
Based on dab0d8e6

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk 2

magicrub, 2017-01-06:
The airspeed sensor does this too, or at least used to until we added sanity checking spike rejection

proficnc, 2017-01-06:
Is this using the new internal drivers? Or the PX4 drivers?

tridge, 2017-01-11:
losing health means no reading for 500ms, which is a very long time on SPI

tridge, 2017-01-11:
looks like this is based on fw from July 2016, so that will be the old px4 drivers. Any chance you could try again with the new driver system in master?

OXINARF, 2017-01-11:
@tridge This is actually from November 28. GitHub loves to show the author date instead of the commit date. In any case, this is a bit old (more than 400 commits since then).

OXINARF, 2017-06-02:
@jschall Have you seen this again? Lots of changes in drivers since then.

jschall, 2017-06-02:
Yes, I see it on Matternet firmware, which uses the ardupilot drivers.

See matternet/ardupilot. It was rebased a couple months ago.

tridge, 2017-06-16:
definitely worth further investigation, adding some perf counters for the various possible error conditions. I'd also suggest making the test faster by making it show unhealthy if it doesn't have a sample for 30ms or so.
I can't think of a reason why this would be happening right now, but some debug code may find it.

peterbarker, 2021-07-30:
So 4 years on, has any further work been done on this?

@jschall did you add any logged diagnostics into matternet's code?

IamPete1, 2021-09-12:
@jschall Bump on Peters question. 

If there are no fresh reports I this we should close.

IamPete1, 2021-10-01:
Any updates from anyone?