ArduPilot/ardupilot #5504: Occasional erroneous values from i2c magnetometer

jschall, 2017-01-05:
#### Issue details
This is a plot over 15 hours:
[image]

There are occasionally these huge outliers from the first magnetometer.

#### Version
Based on dab0d8e

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk 2

gmorph, 2017-01-05:
Hi John.  Its not something weird like someone bumping the table at that time.  I guess we would see shifts in the other axis if that happened.
Hmmm, do you think it would be worth repeating the experiment (I guess it was overnight?) with both a Pixhawk1 and a Pixhawk2 running at the same time and comparing the results.  This way we could rule out anything environmental if we see blips in only one but not the other.
Thoughts?
Thanks, Grant.

gmorph, 2017-01-05:
Did you notice anything else happen in the dataflash logs at the time of the blips?
Thanks, Grant.

jschall, 2017-01-06:
Nothing at the same time.

tridge, 2017-01-11:
i2c doesn't have any checksums, so this could just be bus noise. 
We could try and include some sort of spike filter, but really the EKF is already doing that for us.

tridge, 2017-01-11:
if we double sampled we may be able to detect these, but it would halve the sample rate and that could cause aliasing issues

OXINARF, 2017-06-02:
Can this be closed?

WickedShell, 2017-06-02:
I'm closing it and pointing to the I2C DNF as the fix. I could reproduce this problem before with noise on the I2C bus, https://github.com/ArduPilot/ardupilot/pull/5935 if anyone disagrees this can be reopened, but at a minimum would need to prove that it still happens now after that patch.

closed 2017-06-02 by WickedShell