ArduPilot/ardupilot #5505: Onboard compass call over WiFi/UDP link does not send progress messages

DonLakeFlyer, 2017-01-05:
Using ArduCopter stable 3.4.3 if I do an onboard compass cal over usb connection it all works fine. But if I try the exact same thing over a UDP connection the MAG_CAL_PROGRESS/MAG_CAL_REPORT messages are never sent to the groundstation. The vehicle is in onboard cal mode since I can hear it beeping like it does when doing onboard cal. It just doesn't report progress.

WickedShell, 2017-01-05:
How is the radio connected? I'm assuming it's just a normal mavlink serial
telem as far as the autopilot is concerned?

DonLakeFlyer, 2017-01-06:
It's a PixRacer with the ESP8266 thingy. I assume that is how it is connected. Not much of a hardware person. Phillip Rowse built it for me.

WickedShell, 2017-01-06:
Should be working, will have to check that starting the cal on one port
will still broadcast to all the others, but it should be behaving that way,
I think I had tested this in the past, but I can't recall at this point.

proficnc, 2017-01-06:
Phillip K? Maybe?

DonLakeFlyer, 2017-01-06:
Teehee. Yeah, Thanks to the proficnc guys.

OXINARF, 2017-01-10:
Is this solved? If I heard correctly, @lvale said in the dev call that everything is working.

lvale, 2017-01-10:
can't test again for 3 days, rebuilding racing quad :)

DonLakeFlyer, 2017-01-11:
I just tried it again with the same results. I tried debugging firmware myself but QGC won't run against laster master ArduCopter since it sends MAV_TYPE=0 in heartbeat on flash.

lvale, 2017-01-11:
Just finished testing. with Copter 3.4.3 and QGC initiates onboard cal and on wifi no feedback, while on usb there's the progress bar at the top and when finished the "health" displays.

Tested with latest master
[image]
[compass_cal ardupilot.mavlink.zip](https://github.com/ArduPilot/ardupilot/files/700006/compass_cal.ardupilot.mavlink.zip)

peterbarker, 2017-01-12:
I've just tested with MAVProxy and both Plane and Copter firmwares running over esp8266 worked as expected.

I'm compiling QGC to see how it behaves.

DonLakeFlyer, 2017-01-14:
The real test is with Mission Planner over WiFi. I can seem to get that to work in my Windows VM so I can't test it myself. MAVProxy tends to do strange things since it isn't the same as a direct QGC connection. At least that has been my experience using it with SITL.

OXINARF, 2017-01-22:
I just tested the Bebop2 (which uses UDP over Wifi) with MP and it works.
The Wifi interface in the Pixracer is a simple serial connection so it should work. Can you check your SERIAL5_PROTOCOL and SERIAL5_BAUD parameters? They should be 1 and 921 (or 921600) respectively.

proficnc, 2017-01-22:
Did this on a solo last night, progress bar worked fine on mission planner.

lvale, 2017-01-22:
[image]
[image]
[image]

OXINARF, 2017-01-22:
What is the difference between Bridge/QGC and QGC/Bridge?
Are those packet counters from forever or just since boot? Either way, that's very high packet loss.

lvale, 2017-01-23:
those are cumulative. I should have reset'em before the screenshot, because includes losses from powering on/off the drone

OXINARF, 2017-01-23:
Ok. What is the SR3_EXTRA3 value?

DonLakeFlyer, 2017-01-23:
@lvale Did you try PixRacer+WiFI+MP combination on same vehicle that failed with QGC? I couldn't test that myself. Seems weird it could be QGC only problem, but certainly possible.

lvale, 2017-01-23:
@DonLakeFlyer Same issue with MP latest beta, so it's not a QGC only issue

@OXINARF 
[image]

OXINARF, 2017-01-24:
All looks OK. Without having a Pixracer with a ESP8266 it's hard to help more. Two suggestions I can make are:
- check if you receive MSG_EKF_STATUS_REPORT and MSG_VIBRATION messages
- try to connect a normal radio to the Wifi pins and see if the same happens. If it does, it sure is an ArduPilot issue, if it doesn't, then it is an issue with the ESP8266 firmware

OXINARF, 2017-02-11:
http://discuss.ardupilot.org/t/esp8266-wifi-on-pixhawk/14681/13?u=oxinarf

@DonLakeFlyer @lvale Please try to update your ESP8266 firmware and check if you still don't receive the compass calibration messages.

DonLakeFlyer, 2017-02-19:
I flashed the latest firmware and now it's not working at all. Going to take me a bit to recover from that!

WickedShell, 2017-02-19:
Don, copter, plane or Rover?

DonLakeFlyer, 2017-02-19:
I"m talking about flashing ESP8266 firmware, not ardupilot firmware. For ardupilot I've been testing with Copter.

WickedShell, 2017-02-19:
Who maintains the ESP firmware as it seems like that is potentially the
core problem. (At least I was completing onboard cals with the ESP about 2
weeks ago on a Pixracer)

DonLakeFlyer, 2017-02-19:
You were completing onboard cals with a PixRacer and QGC?

WickedShell, 2017-02-19:
PixRacer and Mission Planner is what I was testing against, since that was
what others on discuss.ardupilot where reporting problems with.(Also passed
my own GCS implementation of compass cal)

DonLakeFlyer, 2017-02-19:
@WickedShell On your GCS do you specific individual compass bits in MAV_CMD_DO_START_MAG_CAL or do you specify all using =0? That's the only possible difference I can think of.

DonLakeFlyer, 2017-02-19:
My call is like this:
```
            _vehicle->sendMavCommand(_vehicle->defaultComponentId(),
                                     MAV_CMD_DO_START_MAG_CAL,
                                     true,          // showError
                                     compassBits,   // which compass(es) to calibrate
                                     0,             // no retry on failure
                                     1,             // save values after complete
                                     0,             // no delayed start
                                     0);            // no auto-reboot
```

Where compassBits has the first three compasses turned on.

WickedShell, 2017-02-19:
I'm not specifying the compasses at all, and allowing it to do all of them:

My call: (clojure, the java bindings allow you to assume that every field
is 0 unless you set it)

  0 (defn begin-compass-calibration

  1   "Requests the start of a onboard compass cal"
  2   [src-msg]
  3   (let [command (new msg_command_long)]
  4     (set! (.command command) MAV_CMD/MAV_CMD_DO_START_MAG_CAL)
  5     (set! (.param2 command) 1) ; retry on failure
  6     (set! (.param3 command) 1) ; automatically save on completion
  7     (log/user "start calibration" src-msg)
  8     (async/>!! radio/transmitQueue (.pack command))))

I can try and test with master again here. I'll need to grab a new version
of QGC and can test that as well. I assume using the daily windows builds
would be fine?

DonLakeFlyer, 2017-02-19:
Could you change your code to pass in the same settings as QGC (3 bits and no retyr) and see if it still works?

DonLakeFlyer, 2017-02-19:
Since I toasted my Pixracer WiFi I'm dead in the water till I figure out how to revive it.

WickedShell, 2017-02-20:
Status update from testing today: With master copter I couldn't begin a compass calibration over ESP8266 with Mission Planner, QGC or my GCS. However I could preform a compass calibration over USB with both MP and my GCS. (QGC was not tested with copter over USB)

With Plane however I could initiate a compass calibration over the ESP8266 on the same version. I'm not really sure what to make of that.

peterbarker, 2017-03-04:
If you could try `026c65d3646b0f5671bf5c5a7d58c5d2cc82844c` then try `026c65d3646b0f5671bf5c5a7d58c5d2cc82844c^` to see if you see the same thing that would be great.

Note you'll want a `git submodule update` between as it is a submodule commit in question.

DonLakeFlyer, 2017-03-04:
FYI: My pixracer WiFi is down for the count right now. I tried flashing a new firmware and it bricked it. Waiting for a breakout board thingy so I can flash firmware over FTDI. Hopefully @lvale can help out with testing until then. I'm probably a couple weeks away from getting my thing from china I need.

peterbarker, 2017-03-04:
@DonLakeFlyer Was your attempted update via the "update" webpage it offers?  That also soft-bricked mine...

DonLakeFlyer, 2017-03-04:
@peterbarker Yes it was. 
@dogmaphobic So not just me with bad experience on firmware flash. LIkely something wrong there.

peterbarker, 2017-03-17:
No worries starting compass calibration now :-)

closed 2017-03-17 by peterbarker