ArduPilot/ardupilot #5506: Copter: mission flies too low perhaps because of ekf-alt vs home-alt error

rmackay9, 2017-01-06:
There's a report on the forums of a vehicle flying about 4m too low during a mission.  Strangely the take-off command gets the vehicle to the correct altitude (10m) but the vehicle then descends about 4m for subsequent mission items.
I've actually seen this behaviour myself on my EnRoute EX700 as well so I suspect we have an issue with a conversion of alt-above home to alt-above-ekf-origin somewhere.

Here's a link to the report on the forums: http://discuss.ardupilot.org/t/latest-firmware-hex-flying-about-4mts-low

OAPpilot, 2017-01-06:
I'd like to clarify that what appears in the CTUN.Alt and CTUN.DAlt fields is the actual and desired altitude above the EKF origin. The EKF origin altitude is normally close to the home altitude but they can be different by a few meters
Hi rmackay9
How do you find the EKF altitude.? THe hex was in launch position when it was powered up for the flight,and home (launch position)  when it was armed.

rmackay9, 2017-01-06:
@OAPpilot, the ekf's alt above sea level is captured in the ORGN,1 message (ORGN,0 is the home location)

IamPete1, 2021-08-30:
I think we should probably close, no new reports 4 years later.

closed 2021-08-31 by rmackay9