ArduPilot/ardupilot #5520: Erratic descent for ArduCopter 3.3.1 & 3.5-dev triggered by simulaneous mode-switch & failsafe

ahkorhonen, 2017-01-07:
#### Issue details
I noticed this while flying a hexacopter with 3.3.1 FW. The issue causes a slow descent even though either AUTO-mode or RTL-mode are on (irrational mode behaviour).

The issue is caused by mode switch to STABILIZE while simultaneously (or after a small delay) the RADIO_FAILSAFE is also activated. From parameters, "Continue Mission in AUTO" was been selected.

After 3.3.3 real life testing, tested the copter with latest dev-version (3.5.0) in SITL. Got the same results (slow descent), but also noticed that if RTL activates instead of AUTO, it will act irrationally as well (slow descent). 

Test setup in VMWare Ubuntu + Mission Planner - controlling via Mission Planner with a joystick with CH3 + CH5 mapped to the same channel. Lowering the channel & then rapidly increasing causes STABILIZE -> FAILSAFE -> AUTO (or nothing for RTL-case). 

1. If STABILIZE + RADIO_FS are activated simultaneously -> RADIO_FS persists for 1s+ in STABILIZE ->RTL triggered -> **RTL starts an "irrational" vertical descent**
(check screenshots in Drive-folder - logs in the same folder)

2. If STABILIZE + RADIO_FS are activated simultaneously -> AUTO-mode on and RADIO_FS turned off within 1s ->AUTO-mode -> **AUTO-mode starts an "irrational" vertical descent**
(check screenshots in Drive-folder - logs in the same folder)

Other modes than stabilize have not been tested as the "mode when failsafe activates". It seems the "key" to the behavior is quick switch to non-"Continue Mission in AUTO" failsafe mode after the initial state.

Mission Planner telemetry shows the following during the case
A. AUTO-mode is on (or RTL in the other case)
B. No failsafes active

The behaviour is similar to having turned LAND-mode on or a failsafe activating. The descent will end and behaviour goes back to normal if mode is changed from AUTO-mode to any other. Turning mode back to AUTO after "any other" will resume mission normally. RTL-mode also functions normally during the case.

The (likely) cause for this to happen in real world was having incorrectly set up RC-receiver failsafe settings (CH3 -> FS level, CH5 -> STABILIZE). The bug was only "realized" along with the incorrect settings due to lost RC-signal timing - as I had a loss of RC-connection of around 1 second (AUTO-mode seemingly on during descent). If the loss would have been longer the mode would have switched to RTL (not tested in RL).

However, in case a radio "glitch" happened to a correctly set up RC-receiver this problem could very well happen (as radios can be more vulnerable to glitching at the "edge" of the radio range).

#### Version
APM:Copter V3.3.1 (real life test) & APM:Copter V3.5-dev (SITL test)

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
This behaviour happens to both QUADs and HEXAs - other multicopters or tradheli not tested.

#### Hardware type
Pixhawk (3DR) & SITL

#### Logs
Logs and screenshots from logs found here:
[https://drive.google.com/drive/folders/0B6HVX1lHgIezd1FBaWNCSC1MS1k?usp=sharing](url)

peterbarker, 2017-03-31:
@ahkorhonen Could you provide a step-by-step list of commands we can enter at the SITL prompt to reproduce, please?

davidbuzz, 2017-04-02:
I haven't looked at the logs, and the report is kinda hard to understand,  but it sounds like the vehicle is triggering some sort of fail-safe event that is causing it to go into "LAND" mode, and the GCS is not receiving any message about it ( or the gcs is recieving the message, but it's quickly being overwritten by further events so the LAND is not apparent to the user ) .   Is this a reasonable description?

ahkorhonen, 2017-04-02:
@davidbuzz Yes, your description matches the events.

OXINARF, 2017-04-02:
The user can always - even during a failsafe - choose to change mode. With a radio failsafe the input from radio should be ignored so you can only change mode with a GCS if your telemetry link is still up. From the description I can't understand if that's what's happening or if it is something else.

@ahkorhonen You said this is was reproducible in SITL, is there a list of steps we can follow to reproduce it?

OXINARF, 2019-08-20:
Closing this as it is old and never got the steps to reproduce.

closed 2019-08-20 by OXINARF