ArduPilot/ardupilot #5521: Quadplane: auto mission when translation to copter it cannot land qhover mode cannot control the attitude

vitsunhoo, 2017-01-07:
#### Issue details
auto mission when translation to copter the qhover mode can not control the attitude
if i do vtol land it won't auto land only hover in 50%throttle ,i switch to qhover doesn't work 
i must switch to qstiblize inorder to land the quad

#### Version
arduplane3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover

#### Airframe type
delta quad plane

#### Hardware type
pixhawk 

#### Logs
http://pan.baidu.com/s/1i4GVmp3

magicrub, 2017-01-07:
Thank you for posting the log, that will be helpful. Can you also post the mission too please?

vitsunhoo, 2017-01-08:
yes ofcourse  there is one of my mission file another i cannot find it
i also switch from FBWA yto QHOVER to priove it doesn't matter with the mission it still cannot control attitude sometimes

magicrub, 2017-01-08:
Are you saying that it can't control it no matter what mode you're in? Have you ran autotune yet?

vitsunhoo, 2017-01-08:
no i didn't  my plane fly very well expect vtol land  i am using a CUAV pixhack v3  today i try again it still can't land normaly

vitsunhoo, 2017-01-08:
i can control well in qstablize and qhover but if i translation to qhover it can't control attitude sometimes
the EK2 IMU MASK is=1

magicrub, 2017-01-08:
You need to do an autotune.

vitsunhoo, 2017-01-08:
autotune the quad?  Does it have this feature?

vitsunhoo, 2017-01-08:
it fly very well if i take off from qhover  mode not translation to quad

vitsunhoo, 2017-01-09:
i try again today .sometimes it will descend when reach the vtol land waypoint . but the descend speed is not
1.5m/s (my Q_WP_DOWN is1.5m/s) it's almost 2.0m/s and it's still increasing. sometimes it will slowly rise up 。
anyway it can't nomaly descend .  the plane will increas 10m when translation to quad.
[image]

IamPete1, 2021-08-15:
sorry we never got to this.

closed 2021-08-15 by IamPete1