ArduPilot/ardupilot #5523: Linux: crash in perf_begin()

tridge, 2017-01-08:
I've had multiple crashes on Disco with the following backtrace:

#0  0x000de154 in Linux::Perf::begin(void*) ()
#1  0x000baa94 in NavEKF2_core::UpdateFilter(bool) ()

I think the problem is a race condition in Linux::Perf::begin(). It de-references the _perf_counters vector without a lock. If this happens while another thread is allocating a perf counter and that re-allocs the array then we can end up referencing old memory. The std::vector class is not thread-safe (as has been discussed before).
We can either add a lock (which could be a performance bottleneck as we don't have RCU) or we could make that vector a fixed length larger than the maximum expected number of perf counters.

Unfortunately I can only reproduce the issue when I don't build with debug enabled. That makes it hard to absolutely confirm the cause, but the symbols in the backtrace do make it seem likely.

I think we need to look over all uses of std::vector to check they are used in a thread-safe manner.

lucasdemarchi, 2017-01-08:
> We can either add a lock (which could be a performance bottleneck as we don't have RCU) or we could make that vector a fixed length larger than the maximum expected number of perf counters.

you can easily test if that's the reason by just increasing the number of `perf.reserve(50)`. Make it stupidly higher like `perf.reserve(5000)` to see if that was the reason.

lucasdemarchi, 2017-01-08:
And we can fail the Perf::add() call in order to avoid the realloc.

tridge, 2017-01-08:
hmm, looking at all the places that use perf_alloc now, I can't see any that could be racing with the EKF call. So I now think this isn't the cause of the crash.
I'm now trying with -O3 and -g to see if I can get a better backtrace. Unfortunately it is the type of bug that can take an hour or two to reproduce. It first happened while I was preparing for a flight this morning, and I've been trying to track it down since then. The above was the first backtrace I managed to get of it.

IamPete1, 2021-09-12:
I wonder if this was fixed by https://github.com/ArduPilot/ardupilot/commit/a63eac8073e7e89d8b0a8da8bce3822abd72be41