ArduPilot/ardupilot #5526: AP_Arming uses AP_Notify to retrieve failsafe state

rmackay9, 2017-01-09:
AP_Notify is meant only to provide notification of the vehicle state to the user so it should be theoretically possible to remove the AP_Notify library and have the vehicle operate normally (i.e. no real functional change beyond the loss of status reporting to the user).
The AP_Arming library breaks this though by using AP_Notify to capture the vehicle's radio (aka throttle) failsafe and battery failsafe status.  bad!
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L371

peterbarker, 2017-12-19:
Easiest fix here is probably to move those checks to AP_Arming_Copter where the arming library has access to the failsafe object.

enjoybeta, 2018-03-28:
https://github.com/ArduPilot/ardupilot/blob/86e495f415d4ddcf7046e7c2bece3d68a1b2c970/libraries/AP_Arming/AP_Arming.cpp#L371

WickedShell, 2018-03-28:
I'm handling the battery one already. (There is an expected follow up PR to the two stage failsafe one that I'm sending in shortly that consolidates the battery information elsewhere).

IamPete1, 2021-08-15:
Battery case is fixed but the RC failsafe still remains,

https://github.com/ArduPilot/ardupilot/blob/0b0ca6a9e99e5f22621af1d0199f008ce1c786d0/libraries/AP_Arming/AP_Arming.cpp#L669

The fix is probably to add a pure virtual AP_Vehicle method to check rc fail safe state.

peterbarker, 2022-07-27:
A PR exists to remove the line from AP_Arming now: https://github.com/ArduPilot/ardupilot/pull/18358

There was some resistance at the DevCall where it was brought up for merge.