ArduPilot/ardupilot #5539: Plane: APM switching from manual to circle mode

GitHubStandub, 2017-01-10:
#### Issue details

When flying in manual mode, the plane suddenly climbed for a while, went down really fast and ended up crashing agains the ground: I could not control it.

So, I took a look at the logs but I still don't know what really happened during that flight: it appears that APM switched to circle mode for 5s and then RTL.

Why would APM switch to circle mode? Is there some kind of failsafe activated? I thought they were all deactivated... (except the throttle failsafe). And of course, why did it crash?

Thanks

#### Version
Based on 7674c60a

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Airframe type
Plane

#### Hardware type
Parrot Disco

#### Logs

mav.parm and flight log file:
[logs.zip](https://github.com/ArduPilot/ardupilot/files/696858/logs.zip)

magicrub, 2017-01-10:
Definitely strange. There is no known mechanism that sets the mode to circle unless a fail-safe is configured for that. 

Something to check: Is circle mode set as one of the 6 modes and is the pwm threshold between modes close enough that noise made it switch modes?

WickedShell, 2017-01-10:
Throttle failsafe while in  manual means go to circle, and then the action system would transition you to RTL after a time delay. I'm not sure whats unexpected here? (Haven't grabbed the logs yet)

peterbarker, 2017-01-11:
On Tue, 10 Jan 2017, GitHubStandub wrote:

> Why would APM switch to circle mode? Is there some kind of failsafe activated? I thought they were all deactivated... (except the throttle failsafe). And of course, why did it crash?

2017-01-10 21:15:22.89: MSG {TimeUS : 313234383, Message : Throttle failsafe on 1513}
2017-01-10 21:15:22.89: MSG {TimeUS : 313234405, Message : Failsafe. Short event on, }
2017-01-10 21:15:22.89: MODE {TimeUS : 313234441, Mode : 1, ModeNum : 1, Rsn : 0}
.
.
2017-01-10 21:15:28.05: MSG {TimeUS : 318395184, Message : Failsafe. Long event on, }
2017-01-10 21:15:28.05: MODE {TimeUS : 318395266, Mode : 11, ModeNum : 11, Rsn : 0}
2017-01-10 21:15:28.05: MODE {TimeUS : 318395269, Mode : 11, ModeNum : 11, Rsn : 0}

What protocol is being spoken between your RC receiver and your autopilot?

tridge, 2017-01-11:
on SBUS (and some other protocols) we also consider it a loss of R/C input if the receiver sets the failsafe bit in the protocol. 
For example, if you configure a FrSky X8R for "keep same output on loss of transmitter" then the values shown in the log will remain the same, but it will go into failsafe as the SBUS protocol will set the "I am in failsafe" bit in the protocol.
I think we should add a parameter for choosing if we should ignore that bit in the protocol

GitHubStandub, 2017-01-11:
Thank you for the replies.

@ magicrub: No, I did not set the circle mode as a flight mode. The closest flight mode to manual is FBWA.

@ peterbarker: I was using Futaba T10J as a transmitter and Futaba R2008SB S-FHSS (SBUS) as a receiver.

I have another question that I forgot to mention: the Disco LED was flashing green quite fast before throttle arming: the code says "fast flashing green if disarmed with GPS 3D lock and DGPS". That never happened to me before. The Parrot Disco board has only one GPS, SO, why was DGPS activated?

Thanks

WickedShell, 2017-01-11:
DGPS can be reported for when you have a fix with a SBAS system, as that is technically a form of DGPS.

The control surfaces (at least pitch) under autopilot control are reversed which is why it crashed when it went to circle/RTL. Circle will always be the immediate action if under manual control and you hit a throttle failsafe.

As @tridge explained the SBUS can report failsafe which is why the aircraft went into the throttle failsafe condition when it got to far away from you that it was starting to lose link.

Closing as this is a support issue, and there aren't any flight related bugs here.

closed 2017-01-11 by WickedShell

peterbarker, 2017-01-11:
Roll also seemed to be reversed.

Your pre-arm routine should include putting your plane into a stabilized mode (FBWA is typical), and checking that the vehicle deflects surfaces to counter any movement from level flight.

The vehicle went into circle because you lost RC to it; it entered throttle failsafe based of the SBUS "I've failed" bit which the receiver set.  You lost RC at the most distant point of your flight - about 200m.  That's too short - you should investigate that.

The vehicle crashed because your control surfaces deflected the wrong way to counter roll and pitch movements.    Considering the angles and lack of airspeed, the vehicle was almost certainly stalled - but it was always doomed.

That's our guess, anyway :-)

peterbarker, 2017-01-11:
On Wed, 11 Jan 2017, GitHubStandub wrote:

> I suspect there is a communication issue between the transmitter and the
> receiver...

Is it possible you've run the transmitter without the aerial, however? 
That is often very, very bad for them output-power wise.

1m away I would expect anything to work, however.  I wonder if it could be 
some sort of encoding problem.