ArduPilot/ardupilot #5544: Plane: Landing permanently disable geofence

WickedShell, 2017-01-11:
#### Issue details
Once you begin a landing sequence the geofence can be disabled ( https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/ArduPlane.cpp#L878 ) but if you abort the landing or change modes for any reason the fence is never re enabled. (This is not related to AP_Landing changes, and is a pre existing bug)

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

peterbarker, 2018-11-23:
@AshleyTR The first thing you need to do is get a repeatable test showing the problem.  Do you have a SITL setup yet?  If not, follow the Wiki to get one.

We'll have to work out what the new desired behaviour is.  @WickedShell will be able to help here, but I'd guess that you would *probably* only re-enable the fence if it was auto-enabled as part of a mission.  

Once you've done an abort you'd probably monitor your altitude, and if you exceed your takeoff altitude you would re-enable the fence.

Thing to watch out for here include latent state - for example, if the vehicle actually lands we probably don't want to magically re-enable the fence if you throw the Plane off again in (e.g.) FBWA and climb above the former takeoff altitude.

We probably need to discuss this on a devcall to make sure the Plane leads (@tridge @magicrub) know what's going on.

magicrub, 2018-11-23:
I should be at the next Dev call. Sorry for missing them lately.

joshanne, 2021-03-24:
As part of #15288 this has been (potentially only half) addressed. When a landing sequence is aborted (due to pilot throttling or issuing a DO_GO_AROUND command), the fence is re-enabled so long as the fence auto-enable condition is set. 

ie. If a fence was auto-enabled on takeoff, and was disabled for a landing, when an abort occurs it enables again once at the takeoff height.

This does *not* address a pilot taking control of the vehicle with a mode change. By the same logic (though I have not tested) if during a takeoff the fence would never auto-enable.