ArduPilot/ardupilot #5546: Copter: MAVLink arming command while armed makes copter impossible to arm

horiuchi4dn, 2017-01-11:
#### Issue details
If copter receives the MAVLink arming command (MAV_CMD_COMPONENT_ARM_DISARM:param1=1) while armed, it never accepts the arming even after disarmed.

To reproduce, assign a joystick button with Arming command on Mission Planner, then tap the button twice (Or double tap the Arm/Disarm button quickly) with armable state.

#### Version
3.4.4

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Hardware type
SITL

rmackay9, 2017-01-11:
I suspect this is a mission planner issue, not an ardupilot issue.  Could you raise this in the mission planner issues list instead?  I'll close this one here if that's ok.
https://github.com/ArduPilot/MissionPlanner/issues

rmackay9, 2017-01-11:
Sorry, I take that back,I've just reproduced it.  sorry and thanks for the report!

rmackay9, 2017-01-11:
Fixed in master.  This will go out with AC3.5 which starts beta testing in a week or two.  Nicely found!

closed 2017-01-11 by rmackay9