ArduPilot/ardupilot #5549: MAG_ENABLE fails to disable compass

budebulai, 2017-01-12:
#### Issue details
Using old version ardupilot firmware, we can use MAG_ENABLE to select if enabling compass, now it fails in 0a18f5e7b2 and others.
With new firmware, when setting MAG_ENABLE to 0, firmware skip compass init, but it will still check compass status when arming motors. 
We can see there are three _state[i].use_for_yaw in parameters, if I choose not to use compass for  vehicle heading, I have to set:
MAG_ENABLE = 0
_state[0].use_for_yaw = 0
_state[1].use_for_yaw = 0
_state[2].use_for_yaw = 0
ARMING_CHECK = -5
boring
So MAG_ENABLE means nothing now.

#### Version
0a18f5e7b2

#### Platform
[  ] All

OXINARF, 2017-01-12:
What firmware is this with? Plane, Copter, Rover?

rmackay9, 2017-01-12:
Yeah, I've been thinking about this MAG_ENABLE parameter on and off.  I'm not surprised if we find out it's meaningless.  When we added the COMPASS_USE parameters we probably didn't take the extra step to remove MAG_ENABLE.

rmackay9, 2017-01-12:
By the way, we have some related issues in our to-do list:
https://github.com/ArduPilot/ardupilot/issues/5454
https://github.com/ArduPilot/ardupilot/issues/5451

I guess the point is that it may or may not be possible to fly without a compass in AC3.4.  I think you could try by setting all the COMPASS_USE parameters to zero, then disable the compass part of the ARMING_CHECK parameter and then see what happens :-).  Certainly loiter won't work of course, but Stabilize, AltHold and ACRO should work.

budebulai, 2017-01-12:
Sorry, I made a mistake.
I forget what I've done to the code, now MP report "PreArm: Compass not healthy" even I set all compass params and arming_check to 0.
I have to check now
Sorry again

budebulai, 2017-01-12:
Thank you, @rmackay9 
Copter 3.5 will stop pixhawk from being armed if compass disabled.
And I comment compass check, pixhawk armed

rmackay9, 2017-06-28:
My main sponsor brought this up today actually.  They found it confusing that to arm and takeoff in stabilize mode (without a compass) they need to leave MAG_ENABLE = 1 and then set the COMPASS_USE, COMPASS_USE2 = 0.
I think we should remove the MAG_ENABLE parameter or move it into the COMPASS library.

OXINARF, 2017-06-28:
@rmackay9 I completely agree that it is very confusing. In my opinion we should just remove it. If users don't want to use compass they disable it with the COMPASS_USE parameters. I think that adding a general compass enable parameter would also be confusing.

rmackay9, 2019-10-01:
This parameter MAG_ENABLE has been moved to COMPASS_ENABLE so I think I'll close this issue.  Hopefully compass-enable does disable all compasses - I think it must.

closed 2019-10-01 by rmackay9