ArduPilot/ardupilot #5550: Resume last waypoint after AUTO mission interrupted

samannoy, 2017-01-12:
#### Issue details
i have been working on my old apm 2.6 for a while with firmware version **arducopter 3.2.1** . While testing the AUTO mission, i found that the arducopter does not support resuming to the last waypoint once the AUTO mission is interrupted, and instead restarts from waypoint 1. Is there a way we can do that(**other than manually using actions tab**)? like, by changing the source code in apm?  or is it necessary to use pixhawk for that? or is there already a solution to that in newer firmware versions?

#### Version
Copter 3.2.1

#### Platform
 Copter
#### Airframe type
Quad and Hexcopter

#### Hardware type
APM 2.6

OXINARF, 2017-01-12:
This is a support question and those should be posted at the forum (http://discuss.ardupilot.org).

You are using a very old version. In any case, the mission can be restarted or resumed. By default it is resumed, so you probably changed the MIS_RESTART parameter.

closed 2017-01-12 by OXINARF

samannoy, 2017-01-12:
Thanks a lot