ArduPilot/ardupilot #5554: Copter: separate RC_FEEL for YAW

robustini, 2017-01-12:
In the release of Copter it would be useful to have separate RC_FEEL for YAW.
In a multicopter the rc roll/pitch feeling is quite similar since almost always the system is balanced, but the yaw might want to set differently.

ChrisBird, 2017-08-26:
It appears that from what I can see the roll and pitch are smoothed out with RC_FEEL but the yaw is using a slew rate limiter for most modes.  So in essence the RC_FEEL has no bearing on the yaw axis for most flight modes.  It does appear to be used on the more autonomous modes such as auto, autotune, circle, flip, guided and rtl from what I can see.

Is the slew rate limiter not enough for this?  When flying in stabilize or poshold the RC_FEEL should have no bearing on the yaw output.

lthall, 2018-01-11:
We now have a seperate rate, expo, and acceleration parameter for yaw. I believe this will address this issue.

closed 2018-01-11 by lthall