ArduPilot/ardupilot #5568: Compile question for vrbrain ArduCopter

nix7965, 2017-01-15:
Hi,

I would like to compile ArduCopter for vrbrain.
For PX4, I succeeded with 'make px4' and 'make px4-v2-quad'

However, it does not work for vrbrain. 
I guess 'make vrbrain' seems right.

Does it really support vrbrain ArduCopter?

rmackay9, 2017-01-16:
We're moving to "waf" especially for the linux builds.  Can you try following the instructions here? 
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

This document is linked from the "Building the Code" wiki page but we haven't created a separate wiki page for this yet. http://ardupilot.org/dev/docs/building-the-code.html

OXINARF, 2017-01-16:
@rmackay9 waf doesn't support VRBrain boards yet.

@nix7965 Are you inside a vehicle directory? You need to be.

nix7965, 2017-01-16:
@OXINARF Could you tell me what is the meaning of "inside a vehicle directory"?

OXINARF, 2017-01-16:
You need to be inside ArduPlane or ArduCopter or APMRover2 directories to run the `make vrbrain` command.

nix7965, 2017-01-16:
@OXINARF  

1. With previous version that I had, I could see  "make: *** No rule to make target `vrbrain'

2. So, I downloaded a new version a few minutes ago, its result was 
Checking connectivity... done.
Submodule path 'modules/waf': checked out '73163ef5c117a51f29546d17c2bb2041aecefcdd'
+ [ -d modules/gbenchmark ]
+ [ -f modules/gbenchmark/.git ]
+ [ -d modules/gtest ]
+ [ -f modules/gtest/.git ]
+ [ -d modules/mavlink ]
+ [ -f modules/mavlink/.git ]
+ [ -d modules/mavlink/pymavlink ]
+ [ -f modules/mavlink/pymavlink/.git ]
+ echo modules/mavlink/pymavlink/.git missing - failed module init
modules/mavlink/pymavlink/.git missing - failed module init
+ exit 1
make: *** [CHECK_MODULES] Error 1

3. So, I manually "git clone"  listed in "mk/check_modules.sh" 
Then, I used "git submodule update --init --recursive" 
and disabled " exit 1"  of "mk/check_module" to avoid the above error.
However, I still saw this error.

/home/ardupilot/libraries/AP_Baro/AP_Baro.cpp: In member function 'void AP_Baro::init()':
/home/ardupilot/libraries/AP_Baro/AP_Baro.cpp:328:10: error: 'PX4_BOARD_PX4V1' is not a member of 'AP_BoardConfig'
     case AP_BoardConfig::PX4_BOARD_PX4V1:
          ^
compilation terminated due to -Wfatal-errors.
make[3]: *** [libraries/AP_Baro/AP_Baro.cpp.o] Error 1
make[3]: Leaving directory `/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/home/ardupilot'
make[2]: *** [/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/home/ardupilot/module.pre.o] Error 2
make[2]: Leaving directory `/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build'
make[1]: *** [/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/home/ardupilot'
make: *** [vrbrain-v51] Error 2

OXINARF, 2017-01-16:
@nix7965 When you say you downloaded, does that mean you didn't use git clone but downloaded a zip? Pretty sure that doesn't work.

nix7965, 2017-01-16:
@OXINARF  
In the above my comment, there are three items.
For 1, I used old version of Ardupilot. 

For 2, Since I used old version, I got the latest version using git clone from here. Then, I executed 'make vrbrain'. But, there were errors without modification.

For 3, To solve problem of 2, I used git clone for submodules that I failed to getting . Then, I executed "git submodule update --init --recursive" at "ardupilot/modules" folder. But, still does not work.

OXINARF, 2017-01-16:
@nix7965 Follow this:

```
git clone https://github.com/ArduPilot/ardupilot.git
git submodule update --init --recursive
cd ArduCopter  // or whatever vehicle you want to build
make vrbrain
```

nix7965, 2017-01-16:
I did 

git clone https://github.com/ArduPilot/ardupilot.git
cd ardupilot                                         
git submodule update --init --recursive
cd ArduCopter
make vrbrain

But, I got this error

/home/ardupilot/libraries/AP_Baro/AP_Baro.cpp: In member function 'void AP_Baro::init()':
/home/ardupilot/libraries/AP_Baro/AP_Baro.cpp:328:10: error: 'PX4_BOARD_PX4V1' is not a member of 'AP_BoardConfig'
case AP_BoardConfig::PX4_BOARD_PX4V1:
^
compilation terminated due to -Wfatal-errors.
make[3]: *** [libraries/AP_Baro/AP_Baro.cpp.o] Error 1
make[3]: Leaving directory /home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/home/ardupilot' make[2]: *** [/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/home/ardupilot/module.pre.o] Error 2 make[2]: Leaving directory/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build'
make[1]: *** [/home/ardupilot/modules/PX4Firmware/Build/vrbrain-v51_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/home/ardupilot'
make: *** [vrbrain-v51] Error 2

OXINARF, 2017-01-16:
Ok, I just compiled over here and got the same error. With all the latest changes to master it's possible some things are broken for VRBrain boards.

scenkov, 2017-04-07:
I get same error, still not working.

lucasdemarchi, 2017-05-19:
We need to get PX4 and VRBrain HALs unified, otherwise it's difficult to maintain... also because we don't build for VRBrain on CI.

khancyr, 2019-04-05:
vrbrain is not on chibios, closing

closed 2019-04-05 by khancyr